procount/pinn #1: v1.9.2 - "Error downloading distribution list from Internet"

schubydoo, 2016-05-26:
Downloaded the latest version of PINN-lite.  Upon booting, I receive "Error downloading distribution list from Internet".  I checked internet access with 'online help' and it seems to be working ok.  Also I get the following list automatically: Raspbian, OpenELEC, Data Partition, OSMC, and Windows 10 IoT.  I believe the error stems from loading the alt OS list for Arch (which is exactly what I wanted to use this for).

Version: PINN-Lite v1.9.2
Model: Raspberry Pi 3 Model B
Network: Ethernet (DHCP)

Thoughts?

schubydoo, 2016-05-26:
I forgot to follow the troubleshooting instructions.  I'll grab some logs shortly.

schubydoo, 2016-05-26:
I don't see anything that calls out to me, but I attached the logs.

[debug.txt](https://github.com/procount/pinn/files/283485/debug.txt)
[dmesg.txt](https://github.com/procount/pinn/files/283484/dmesg.txt)

procount, 2016-05-26:
Your logs look fine, but for some reason your Pi cannot download the alternative distribution list. I just tried it on mine and it downloaded ok and I can see it on github too.
The file to download is specified in recovery.cmdline. You could check if this file has not changed. It should include:
`alt_image_source=https://raw.githubusercontent.com/procount/pinn-os/master/os/os_list_v3.json`
If that still looks ok, see if you can download it normally through your web browser.
Or if you can get to the PINN rescue shell (ctrl-alt-f2, username=root, password=raspberry), try downloading it using wget.
```
cd \tmp
wget https://raw.githubusercontent.com/procount/pinn-os/master/os/os_list_v3.json
```
Let me know how that goes. If you still have no joy there is something else to try...

schubydoo, 2016-05-26:
Ok I double checked recovery.cmdline and the URL seems fine.  I'm also able to use wget to download the .json file.  Viewing the file it looks fine as well.  Not sure why it can't download it on startup...

procount, 2016-05-26:
Hmmm... Interesting. The only difference between downloading the alternative file and the standard one is the URL itself, so I think it is a networking access issue of some sort. Where in the world are you?
We can go 1 of 2 ways (or maybe both: 1 then 2):
1) If you are willing to help debug this issue, I will build a version with more debug information to determine why it is not working.
2) We can bypass the json file altogether and get PINN to install it directly from Arch website.

schubydoo, 2016-05-26:
I'm located in Washington state.  I'd be happy to help debug it for you in case anyone else runs into the issue.

procount, 2016-05-26:
Great! Thanks. 
I will add some debug info so we can track it down, but I won't be able to upload it until tonight (UK time).
In the meantime, could you go to [My Profile](https://www.raspberrypi.org/forums/memberlist.php?mode=viewprofile&u=74876) and drop me a PM so I can contact you with how to download the debug version?

schubydoo, 2016-05-26:
I created an account: "Schuby" (https://www.raspberrypi.org/forums/memberlist.php?mode=viewprofile&u=192364)

Unfortunately I cannot send a PM until I'm established apparently. Maybe it will let you PM me?

schubydoo, 2016-05-26:
I got your PM but I can't reply :/. I can email you if you want to provide your email address.

procount, 2016-05-28:
Bug tracked down to the use of https and time setting. Fixed in 1.9.3

closed 2016-05-28 by procount