procount/pinn #2: Add support for gamepad selection of OS

colinchilds, 2016-06-12:
For those of us who have Retropie installed, it would be awesome to be able to select the OS to boot into using the gamepad already plugged into the device, rather than having to plug in a keyboard or mouse.

procount, 2016-06-13:
Thanks for the suggestion

Norde, 2018-09-17:
This would be a very nice enhancement !

This requires additional kernel modules (like xpad for xbox controler).
Does addition of such modules is conceivable for PINN?

procount, 2018-09-17:
PINN does not use a modular kernel, so it's a bit problematic at the moment. But I hope to be able to do something about it at some point, just not immediately.

Norde, 2018-09-17:
What, not immediately ??? 😡  :smile:

Effectively, without modular kernel, there's no easy implementation possible. Do you know some alternative/ tricky way ?

By the way, thank you for all your work on PINN!

andrebadaia, 2019-07-29:
I am interested in this solution too! Maybe you can instruct us to recompile PINN for that support! Thanks!

procount, 2019-07-29:
PINN now includes modular kernel support, so if you have some drivers for your gamepads (preferably in source format) I can take a look at what is needed to include them

andrebadaia, 2019-07-29:
It's awesome!!! I have various types of joysticks, but the XBOX 360 is OK. It uses the xpad driver on linux, isn't?

procount, 2019-07-29:
I don't have any gamepads to test his feature, so I'm going to have to rely on everyone else to provide feedback on this.
I've added an xpad and joystick driver to PINN and uploaded a test version to
https://sourceforge.net/projects/pinn/files/testing/joystick-33d.zip/download .

If you want to test it, please download joystick-33d.zip and install it to a blank FAT32 formatted SD-card just as you would pinn-lite.zip.

I haven't added any support into PINN to actually use the driver yet, so at this stage it's just a question of testing whether the gamepad can be detected and is usable.

Please boot PINN with your USB gamepad attached.
Once booted, drop into the recovery shell by pressing the ctrl-alt-f2 combination and login with the username/password of root/raspberry.
At the prompt type 
```
jstest /dev/js0
```
Make a note of any text that is displayed.
Press the buttons on your joystick and make a note of any messages that are created.

andrebadaia, 2019-07-29:
It's worked! The PINN detects the joystick correctly, and all axes and buttons were detected by jstest on /dev/input/js0. The variations of axes go to -32768 to 32767, and buttons, switch between on and off. Thanks a lot for the effort!!

procount, 2019-07-29:
@andrebadaia Could you please post some output from jstest so I can see what it produces?
What do you expect the joystick buttons to do in PINN?

lurch, 2019-07-29:
&lt;tongue-in-cheek&gt; Can you add a game of Tetris to PINN, so that we have something to do while waiting for the OSes to install? &lt;/tongue&gt; :wink:

andrebadaia, 2019-07-29:
It is equal to Linux, the same buttons and axes in the same place. I'll map for you:

Axis 0: Left Analog X 
Axis 1: Left Analog Y
Axis 2:  LT 
Axis 3: Right Analog X
Axis 4: Right Analog Y
Axis 5: RT
Axis 6: D-pad X
Axis 7: D-pad Y
Button 0: A
Button 1: B
Button 2: X
Button 3: Y
Button 4: LB
Button 5: RB
Button 6: Back
Button 7: Start
Button 8: Mode
Button 9: Left Analog Click
Button 10:Right Analog Click

[image]

About the use, I expect the behavior is the same when the package xserver-xorg-input-joystick of X, when the left analog controls the mouse, A and B are the left and Right click. The other buttons have other funcions, but the minimal usability is enough (just for select OSes).

procount, 2019-07-29:
> Can you add a game of Tetris to PINN

What? Tetris? What's wrong with Minesweeper? 😉

lurch, 2019-07-29:
> but the minimal usability is enough (just for select OSes).

Presumably the minimal usability for just selecting which OS to boot is UP, DOWN and SELECT ?
Might be hard to implement xserver-xorg-input-joystick, given that PINN doesn't use X...

procount, 2019-07-29:
I'll probably do something like duplicating the CEC remote functionality

andrebadaia, 2019-07-29:
In fact, the joystick substitutes the mouse on X, but you can improve the funcionality associating some important functions to the buttons and axes. It's not like a keyboard, but helps a lot!!! I use PINN with Raspbian, Retropie and OSMC, connected on my TV, and all controlled by joystick (I use a virtual keyboard on Raspbian, together with xserver-xorg-input-joystick), but I need the keyboard to choose the OS on PINN.

procount, 2019-08-01:
I have made a first attempt in https://sourceforge.net/projects/pinn/files/testing/joystick-33e.zip/download 
It compiles, but I've done NO testing whatsoever, so it's probably very buggy at the moment. 
If you have time to try it out and feed back I would appreciate it.
I just mapped left analog to the mouse cursor, A=left click, X =Cancel (Esc).

andrebadaia, 2019-08-01:
Yes, it didn't work. The gamepad does nothing on GUI (it continues working on jstest).

procount, 2019-08-01:
Sourceforge has been a bit slow and hasn't indicated anyone had downloaded the file, so I was lazy and pushed up several versions of the same file with various fixes in it. The early ones wouldn't have worked but I thought the last one would have worked. 
Can you check you have the latest one? I pushed it up at 15:16 (BST)

andrebadaia, 2019-08-01:
I tried again (I've downloaded the last one version presented in sourceforge files) and it doesn't work too.

procount, 2019-08-01:
OK thanks. I'll create a debug version to display some traces in /tmp/debug and we'll see if we can find out what is going on.
Out of interest, does that file contain anything related to the joystick already?

andrebadaia, 2019-08-01:
Sorry, I don't understand your question. What do you mean?

procount, 2019-08-01:
Once booted, drop into the recovery shell by pressing the ctrl-alt-f2 combination and login with the username/password of root/raspberry.
At the prompt type
`less /tmp/debug`
Browse using the cursor keys to see if there is any mention of the joystick.
Press 'q' to quit.

You could also try 
`tail -f /tmp/debug`
And then press the joystick buttons and see if there are any messages displayed.

procount, 2019-08-01:
I pushed up joystick-33f.zip with more trace & debug messages.

andrebadaia, 2019-08-01:
less /tmp/debug

It recognizes the joystick (Microsoft X-BOX 360 pad - 8 axes, 11 buttons), load CEC mappings from ":/cec_keys.json" and load joystick mappings from ":/joy_keys.json". No error reported.

Pressing the buttons and axes didn't make any difference.

andrebadaia, 2019-08-01:
Where are the files cec_keys and joy_keys.json? I didn't find them on PINN or SETTINGS partition...

procount, 2019-08-01:
Is that still with v3.3e or the new v3.3.f?

andrebadaia, 2019-08-01:
3.3e. I'll try the 3.3f.

procount, 2019-08-01:
> Where are the files cec_keys and joy_keys.json? I didn't find them on PINN or SETTINGS partition..

https://github.com/procount/pinn/blob/master/README_PINN.md#cec-control

andrebadaia, 2019-08-01:
Do I have to create these files?

procount, 2019-08-01:
No. Not for this testing - it will just complicate matters. I was just answering your question.

andrebadaia, 2019-08-01:
The /temp/debug output is here (3.3f):

[image]

procount, 2019-08-01:
It would be more nteresting to see the log after that, after you have pressed all the buttons on your joystick.

To capture the debug logs you can do the following: Press Ctrl-Alt-F2 to get to the recovery shell. Use the username:root password:raspberry.

```
cd \tmp
mount -o remount,rw /mnt
cp debug /mnt/debug.txt
sync
```
Now you can turn off the Pi, put your SD card in a PC and copy the debug.txt file to your PC.

andrebadaia, 2019-08-01:
That's it.

[debug.txt](https://github.com/procount/pinn/files/3457988/debug.txt)

procount, 2019-08-01:
Thanks. Interesting. I assume you pressed joystick buttons during all of that debug capture, so as nothing is being detected I shall have to think again.

andrebadaia, 2019-08-01:
I did the test again, to be sure... I pressed all gamepad buttons for almost 1 minute. 
[debug.txt](https://github.com/procount/pinn/files/3458387/debug.txt)

procount, 2019-08-01:
Try Joystick-33g.zip

andrebadaia, 2019-08-02:
It's a beginning! The mouse pointer started to move in g revision!

Reports:
The Y axis is inverted;
The moviment is not fluid, seems like jumps... If you try to hold the axis, the pointer goes crazy!
The A button is ok, I could navigate on all the PINN windows with it;
The X button closes the window - if you press it on boot window, the PINN reboots.

Well, that's it! You are on the right track!

[debug_g.txt](https://github.com/procount/pinn/files/3459411/debug_g.txt)

procount, 2019-08-02:
Yay! That's mostly as expected.
The mouse movement will never be smooth like a proper mouse, but I can improve it.

> f you try to hold the axis, the pointer goes crazy!

I'm not sure what you mean by that?

andrebadaia, 2019-08-02:
When you hold the axis (for the left, for example), the pointer goes to other position, down, or up! You have to "press and leave" the axis to get the right moviment.

procount, 2019-08-03:
Can you try joystick-33h?
I've added some more buttons etc. Try the left and right analog, plus D-pad.
The right  analog joystick should behave differently and have some acceleration which you can alter by modifying joy_keys.json where you can also see the mappings I added in joy_keys.json. This is similar to the cec_keys.json and documented in the previous link I sent you: https://github.com/procount/pinn/blob/master/README_PINN.md#cec-control
You should be able to map any button onto any of the available keys.
Please try it out and experiment with the key buttons to see which works best with the joystick.

procount, 2019-08-18:
Do you have any more feedback on this? Does it work better now?

andrebadaia, 2019-08-19:
I am sorry, I am back to work and I had no time to keep testing...

Norde, 2019-09-05:
I'll be able to test this during october :)

Just out of curiosity, what prevents / makes complicated the support of gamepad via bluetooth?

procount, 2019-09-05:
Mainly bluetooth itself.
NOOBS & PINN are designed to be quite small and the bluetooth stack can be quite large.
There are also complications with having to pair devices (and remember them).
I intend to look at it at some point, as I have a bluetooth keyboard I would like to use with PINN myself, but I have more pressing issues to look at first.

Norde, 2019-09-05:
Ah yes I understand, a lot of work for a non-essential function ^^

Anyway thank you for all the work done on Pinn!

RaspberryPiFan, 2019-11-07:
@Norde Did you test it?

procount, 2019-11-22:
Joysticks and gamepads supported in v3.3.2

closed 2019-11-22 by procount

Norde, 2019-12-21:
Sorry, I'm arriving a little (a lot) late for the tests.

I'm using an Xbox one controller (rev 1708) with a rpi3B +.

Something strange happen with pinn 3.3.2 (and joystick-33h.zip) :
My controller is detected (in /dev/input/js0).
However the controller seems not "paired" (the home button of the controller light up when pairing is ok, if not controller is just charging).

Most of time light stay off (and not input is detected).
In rare case controller pair well after random time (light is up and inputs works) and launching jstest (on /dev/input/js0) force this pairing and then controller works.

And there's nothing in the debug log about this (only "loading CEC and joystick mapping" when pairing is ok).

Some clue about this strange behavior ?

procount, 2019-12-21:
What do you mean by "paired"?
PINN dies not support Bluetooth, only USB, and there Is no pairing in USB.

Norde, 2019-12-21:
I'm using USB of course.
I say "paired" in oposition of "charging" (Xbox One controller has a battery), i misspoke.

The gamepad is recognized under /dev/input/js0 but only charging (input are not listened), except in rare random cases (that I can't reproduce) or when i'm listen with jstest wich seems force the "input listening state".

procount, 2019-12-21:
Sorry, it's been a while since I did this so I have to remember.... :wink: 
Did I put jstest in PINN? Or are you using it in Raspbian?
If the xboxdrv detects your gamepad it should announce the type and capabilities in the debug log.

Norde, 2019-12-21:
I understand, no problem :)
But the fact that I misspoke didn't help ^^'

You've put jstest in joystick test release (33h here).
In debug log only a "No joystick found" appears. Yet my gamepad device entry is under /dev/input/js0.
When using jstest nothing more appears in the debug.

lurch, 2019-12-22:
If the controller has a battery, then I'm guessing it's a wireless controller, which (presumably) connects to a special receiver USB dongle? I know noting about Xboxes :wink: 

Perhaps there's a delay between the controller and/or the dongle powering up (you mentioned "not paired" earlier), and this delay is causing PINN to not see any joysticks when it tries looking for them? :man_shrugging: :man_shrugging:

Norde, 2019-12-22:
There's no special usb dongle. The controller has  native bluetooth and a usb port for wire connexion.

But you're right, there's a little delay between cable connection/powerup of the controller and the "ready status".

procount, 2020-01-08:
What is the output from `lsusb` when your joystick is attached?

Norde, 2020-01-08:
`Bus 001 Devices 005: ID 0a12:0001
Bus 001 Devices 001: ID 1d6b:0002
Bus 001 Devices 009: ID 045e:02ea <- joystick (xbox one s controller)
Bus 001 Devices 003: ID 0424:2514
Bus 001 Devices 002: ID 0424:2514
Bus 001 Devices 007: ID 0424:7800
Bus 001 Devices 004: ID 046d:c316`

I've the got same output wether joystick is "charging" or "connected" (obtained after running jstest) state (only joystick Devices number change).

lurch, 2020-01-08:
I think the Device number changing indicates that (from the Linux-side of the USB connection) the device appears to have been removed and reconnected (i.e. it's re-enumerated itself when the jstest switches its state).

It might be interesting to get the `lsusb -v -d 045e:02ea` output in both the "charging" and "connected" states?

procount, 2020-01-08:
You'll get more comprehensive results from `lsusb` in Raspbian. The PINN version is rather limited.
But it's enough to confirm that your controller should be supported:
> 0x045e, 0x02ea, "Microsoft X-Box One S pad", 0, XTYPE_XBOXONE 

So maybe it is a timing issue. I'll have to try re-establishing the joystick later on in the process, somehow....

Norde, 2020-01-09:
I can't reproduce this bug under raspbian (the xbox one s controller is directly recognized, never only in "charging" state).

And I don't have enough knowledge yet to properly add a complete lsusb/usbutils to Pinn's buildroot, sorry :/

[12 more replies]