procount/pinn #5: Feature request: Boot pinn-noobs form an USB stick (USB only environment)

franzgl, 2016-09-15:
Working with an USB stick only is really nice.

The posts https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md
and https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=44177 are great.
Those posts descripe USB only booting.

So booting and working with an USB stick is much more fun.

It is the only way to boot a USB storage device without an SD card. When the boot is complete you can insert the SD card. This is nice, because then you can modify the file system or other things that you could not when the root file system is in use.

So here is my feature request:

Please port pinn-NOOBS to an USB stick only environment. USB sticks are more reliable and cheaper than SD cards. I tried to move NOOBS to the USB stick. The boot failed because the file **recovery.elf** has hard coded **/dev/mmcblk0** in it and does not know how to handle **/dev/sda**. I think also **recovery7.img** has to modified.

I hope you will soon release a new **recovery.elf** file that will work with USB only situations.

As I can see you do support USB.  But there are only installation images from an USB stick supported, but not booting from USB.

I asked this issue at https://github.com/raspberrypi/noobs/issues/356 and I do not know if the noobs team is interested.

procount, 2016-09-15:
As I hinted at in response to your post on the forum, I'm sure this is currently being worked on in NOOBS, but it is a complicated enhancement so we need a little patience - it will not happen overnight,
Firstly, as Lurch points out, this is only applicable to the RPi3 since it is only in the new BCM2837 soc. Inline with the RPF's normal goals, I assume they will still only want 1 version of NOOBS that works on ALL RPIs so it must support the existing SD cards as well as the new USB sticks.

Maybe you also expect that NOOBS running on an SD card is able to install OSes to a USB stick, both in USB boot mode for the RPI3, and maybe just for the rootfs for other models? Or NOOBS running on a USB stick to be able to set up an SD card? There are constraints with the boot selection in that NOOBS can only select a boot partition on the same device on which NOOBS is running.
 
As you point out, PINN already supports USB as a source medium, so to add it also as a destination medium will also require some thought so that I can manage 2 USB devices.

So, there are a lot of complications that need to be sorted out, but it will come. We need a little patience. Bear in mind that this USB boot mode, is still only beta and has not hit the main distribution yet.

procount, 2017-04-23:
Available in PINN v2.4

closed 2017-04-23 by procount