procount/pinn #7: Error with CEC compiling

mortaca, 2017-01-05:
`In file included from mainwindow.cpp:17:0:
ceclistener.h: En la función miembro static ‘static int CecListener::clearKeyPressed()’:
ceclistener.h:44:48: aviso: no hay una declaración de devolución en la función que no devuelve void [-Wreturn-type]
     static int clearKeyPressed() {keyPressed=0;}
                                                ^
In file included from mainwindow.cpp:1:0:
mainwindow.h: En el constructor ‘MainWindow::MainWindow(const QString&, QSplashScreen*, bool, QWidget*)’:
mainwindow.h:54:28: aviso: ‘MainWindow::_netaccess’ se inicializará después de [-Wreorder]
     QNetworkAccessManager *_netaccess;
                            ^
mainwindow.h:51:18: aviso:   ‘QMessageBox* MainWindow::_displayModeBox’ [-Wreorder]
     QMessageBox *_displayModeBox;
                  ^
mainwindow.cpp:92:1: aviso:   cuando se inicializó aquí [-Wreorder]
 MainWindow::MainWindow(const QString &defaultDisplay, QSplashScreen *splash, bool noobsconfig, QWidget *parent) :
 ^
mainwindow.cpp: En la función miembro ‘void MainWindow::onKeyPress(int)’:
mainwindow.cpp:2009:10: error: ‘CEC_User_Control_Left’ no se declaró en este ámbito
     case CEC_User_Control_Left:
          ^
mainwindow.cpp:2013:10: error: ‘CEC_User_Control_Right’ no se declaró en este ámbito
     case CEC_User_Control_Right:
          ^
mainwindow.cpp:2017:10: error: ‘CEC_User_Control_Up’ no se declaró en este ámbito
     case CEC_User_Control_Up:
          ^
mainwindow.cpp:2021:10: error: ‘CEC_User_Control_Down’ no se declaró en este ámbito
     case CEC_User_Control_Down:
          ^
mainwindow.cpp:2025:10: error: ‘CEC_User_Control_Select’ no se declaró en este ámbito
     case CEC_User_Control_Select:
          ^
mainwindow.cpp:2038:10: error: ‘CEC_User_Control_Number0’ no se declaró en este ámbito
     case CEC_User_Control_Number0:
          ^
mainwindow.cpp:2041:10: error: ‘CEC_User_Control_Exit’ no se declaró en este ámbito
     case CEC_User_Control_Exit:
          ^
mainwindow.cpp:2044:10: error: ‘CEC_User_Control_ChannelUp’ no se declaró en este ámbito
     case CEC_User_Control_ChannelUp:
          ^
mainwindow.cpp:2047:10: error: ‘CEC_User_Control_ChannelDown’ no se declaró en este ámbito
     case CEC_User_Control_ChannelDown:
          ^
mainwindow.cpp:2051:10: error: ‘CEC_User_Control_F2Red’ no se declaró en este ámbito
     case CEC_User_Control_F2Red:
          ^
mainwindow.cpp:2057:10: error: ‘CEC_User_Control_Number1’ no se declaró en este ámbito
     case CEC_User_Control_Number1:
          ^
mainwindow.cpp:2060:10: error: ‘CEC_User_Control_Number2’ no se declaró en este ámbito
     case CEC_User_Control_Number2:
          ^
mainwindow.cpp:2063:10: error: ‘CEC_User_Control_Number3’ no se declaró en este ámbito
     case CEC_User_Control_Number3:
          ^
mainwindow.cpp:2066:10: error: ‘CEC_User_Control_Number4’ no se declaró en este ámbito
     case CEC_User_Control_Number4:
          ^
mainwindow.cpp:2069:10: error: ‘CEC_User_Control_Number5’ no se declaró en este ámbito
     case CEC_User_Control_Number5:
          ^
mainwindow.cpp:2073:10: error: ‘CEC_User_Control_Number9’ no se declaró en este ámbito
     case CEC_User_Control_Number9:
          ^
make[1]: *** [mainwindow.o] Error 1
`

procount, 2017-01-05:
Good to know someone is using my code!
Looks like I missed out a header file (?)
I'll investigate for the next release.

procount, 2017-01-05:
Is this your first build of PINN, or have you built it (or NOOBS) before?
You might want to try building the whole thing again including the rpi-firmware package. You need to have the vc_cec.h file

mortaca, 2017-01-05:
Is my first build, i put the RPi firm on /buildroot/boot/ ?

Im using PINN to install Recalbox over CRT TV with DPI output

procount, 2017-01-08:
It is a dependency problem.
A workaround is as follows:
```
cd buildroot
make rpi-userland
cd ..
./BUILDME.sh
```
I'll correct it in the next release

procount, 2017-01-10:
Fixed by release p2.1.2

closed 2017-01-10 by procount

phmio, 2017-09-23:
Strange procount,  i have try to build PINN too (from pinn-master) to understand how buildroot works and have got yet the same error on  CEC_* :worried:
I had to apply your workaround.
>bootselectiondialog.cpp: In member function ‘void BootSelectionDialog::onKeyPress(int)’:
>bootselectiondialog.cpp:429:10: error: ‘CEC_User_Control_Select’ was not declared in this scope
>     case CEC_User_Control_Select:
>...
>Makefile:585: recipe for target 'bootselectiondialog.o' failed
>make[1]:** *** [bootselectiondialog.o] Error 1