procount/pinn #8: Please add PiCorePlayer

marakasmalan, 2017-01-05:
Thank you - worked nicely.

Can you please add the option to install PiCorePLayer?

https://sites.google.com/site/picoreplayer/home
https://sourceforge.net/projects/picoreplayer/?source=directory

procount, 2017-01-31:
To include PiCorePlayer, the maintainers will need to provide it in a format that can be installed by NOOBS or PINN. i.e. a name.tar.xz file for each partition, or an .img.xz file for each partition, rather than an image for the whole SD card. (instead of xz compression, zip, gzip, bzip2 or lzo compression can be used) details can be found in the NOOBS readme.md file, or PINN readme_pinn.md file. Also a suitable icon and slide show images would be very useful.

marakasmalan, 2017-01-31:
I am a novice, but sure I will investigate - The installed image is very
small. The icon and slide show images I must be able to help with.

So just an image of the boot partition and an image of the partition where
picoreplayer is loaded? Then, how should I get it to you?

procount, 2017-01-31:
There are detailed instructions on how to create a .tar.xz image here -> https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version
A .img.gz file is very similar, but uses dd instead.

The images need to be hosted on the internet somewhere, like sourceforge or github, where they can be downloaded by anyone. Have a look at an existing image for the support files that need to be added. I can help creating the json files once the images are available.

Sorry, but I don't have the time/resources to maintain and update the images of the OSes myself, but if someone is prepared to do it, I can reference them from PINN.

marakasmalan, 2017-01-31:
> md#how-to-create-a-custom-os-version

For sure, it is the least I can do, I will do it over the weekend.

marakasmalan, 2017-01-31:
RTFM and it shouldn't be that difficult!

procount, 2017-03-13:
I have actually created the appropriate tar.xz files for picoreplayer and picoreplayerAudio (the optimised audio version). The normal version seems to load ok. The optimised version doesn't seem to find any sound  drivers on my PI. Maybe it needs a real HAT or something to work properly?
Anyway I could do with a png icon for it (40x40) and 1 or more slides (387x288) for the installation slideshow, although not many because it installs so quickly, it being so small!

procount, 2017-09-26:
PiCorePlayer is now included.

closed 2017-09-26 by procount

godzinilla, 2022-01-26:
Hello @procount. Would you be so kind and create an updated image for piCorePlayer 8.1.0 64 bit for RPi 4B?

The image available on the PINN server is not booting at all o the RPi4B. Tomorrow I will try with RPi3B.