procount/pinn #10: Update languages used by PINN project (based on NOOBS languages)

bovirus, 2017-01-13:
Hi.

I rebuild the NOOBS project regarding the languages.
in original NOOBS project (then in your project because use the same language files) there are some incomplete language files and many languages proposed by user but without any answer by NOOBS admins.
I completed/aligned/repacked all languages available. Here

https://github.com/raspberrypi/noobs/issues/364

you can find all languages align to master English languages (147 strings).

Please take care also about the necessary changes for Korean language 8add fonts and text)

locks like that are a sign it is missing a font file with the right characters.

**** Korean patches info start

Make sure you do not just copy the .ts file, but apply all changes in the PR including the ones that cause the extra .ttf font files to be added at the right locations: https://github.com/raspberrypi/noobs/pull/199/files

One way to do that is to download all changes as .patch file: https://github.com/raspberrypi/noobs/pull/199.patch

**** Korean patches info end

Could you please update PINN using all languages available (and updated) got from my kit and include the changes for Korean language in the code?

Thanks.

procount, 2017-01-13:
I have been following that issue on Noobs. It will be easier for me to include it once it has been updated in Noobs, rather than doing the work twice.

bovirus, 2017-01-14:
@procount 

Using teh command

ubuntu@ubuntu:~/Desktop/pinn$ lupdate recovery/recovery.pro -no-obsolete -disable-heuristic similartext

I got the error message

Could not find qmake configuration directoryCould not find qmake configuration fileUsing OS scope before setting MAKEFILE_GENERATORProject MESSAGE: Disabling CEC support for Raspberry Pi - rpi-userland headers not found at ../../staging/usr/home/ubuntu/Desktop/pinn/recovery/recovery.pro(25): Function 'system' is not implemented

I found that your translation contrains more strinfgs (184) than NOOBS (147).

then for the reason that I explained befroe (different struccture/different language files) I believe that you can develop the project without waiting NOOBS updates.

s8321414, 2017-01-15:
hmm, so which file should I download to translate?

procount, 2017-01-15:
@bovirus - thanks for doing this work. The translation side of QT is all new to me, but it wiuld be nice to include this.

To fix the CEC error abive you need to build rpi-userland. Just build all of PINN first or see workaround in #7.

Are your translations based on p2.1.2? Is there any possibility to do a pull request instead of posting patches? It is easier to integrate the changes this way.

bovirus, 2017-01-15:
@s8321414 c

You can find the languages file in 

PINN langauge files (Rxx) - https://drive.google.com/open?id=0ByL-fx0GbfuidTM5WjIxUjlGQlE

The additional strings that you should translate is about 36.
Thanks.

bovirus, 2017-01-15:
@procount 

I tried to do

cd buildroot
make rpi-userland

the answer is

make: *** No rule to make target 'rpi-userland'.  Stop.

bovirus, 2017-01-15:
@procount 

The translation files is based on PINN 2.1.2.
I got also the latest source by GitHub reopository.

procount, 2017-01-15:
The CEC message is a compile time message, not a user displayable message, so I don't think it is necessary to translate that.

bovirus, 2017-01-15:
@procount 
I didn't have a big knowledge about repository and make pull request
Ina  few days I hope that I will receive new language updates then if you can you can get download and aply it to your repository.

bovirus, 2017-01-15:
@ aldatsa 
Could you please update Basque translation?

@s8321414 
Could you please update Taiwanese translation?

@Akkiesoft
Could you please update Japanese translation?

I'm sorry but you are the only translators that helped me in NOOBS project to complete your translations.

s8321414, 2017-01-15:
OK, I will do it recently when I have free time.

aldatsa, 2017-01-15:
I have updated the Basque translation. It's attached to this message
[translation_eu.zip](https://github.com/procount/pinn/files/706638/translation_eu.zip)

aldatsa, 2017-01-15:
I created a pull request for the Basque translation as well: #11

bovirus, 2017-01-15:
@procount 

Pull request: https://github.com/procount/pinn/pull/12

s8321414, 2017-01-16:
@procount 
[translation_zh_TW.zip](https://github.com/procount/pinn/files/708302/translation_zh_TW.zip)

procount, 2017-01-16:
I added most of these translations and put a beta version on sourceforge for testing.
https://sourceforge.net/projects/pinn/files/testing/    pinn-1212a.zip
The additional Korean fonts are quite big so it is a bigger download.
Still basque, zh_tw and IT translations to add......

Thanks to everyone who contributed to the translations. 
I would add that I rely totally on your support for these and there is no way for me to validate them all. Therefore I accept no responsibility whatsoever for any incorrect translations.

bovirus, 2017-01-16:
@procount

Thanks for the updates.

Of course you are ABSOLUTELY any responsability for all translations.

bovirus, 2017-01-16:
@procount 

Sorry but I didn't understand. Are available

**recovery/translation_ca.ts
recovery/icons/ca.png**

but is not available

**recovery/translation_cs.ts**

procount, 2017-01-16:
Sorry, my typing error!
I have no recovery/translation_ca.ts file (catalan ?) to match the ca.png file. Should there be one?

procount, 2017-01-16:
It's OK, I found a copy in your bovirus-1 branch

bovirus, 2017-01-16:
@procount 
Please take care that you will find always all files in first post of this discussion.

Question1: where the PINN program get the language name? Spanish (es) and Basque (eu) language fiels seems have the same language description in drop down list in main PINN window.

Question2: which is the file that contains WiFi settings (to avoid to overwrite it during different build test?
WPS

Question3: could be possible to add "Build xxxx" (main window) in language file (to have it multilanguage) and with a european date like dd.mm.yyyy?

procount, 2017-01-16:
@bovirus, I compared the files you posted in the first post of this discussion with those in the pull request. There are some minor differences that might have resulted in minor problems.

**Translation_it.ts:**
Needed = Necessario or **Necessari** ?
Available = Disponibile or Disponibili ?

**Translation_eu.ts:**
language=eu or eu_ES ?

Q1: I was wondering that myself. Maybe changing the language to "eu_ES" will fix it? I will try in next release.

Q2: /settings/wpa_supplicant.conf

Q3: I will try.....

bovirus, 2017-01-17:
@procount 
Thanks.

The right term ise "Necessario" and "Disponibile" because are related to Spazio (Space) that is singular.

Q1: please test as eu_ES.

Q2: thanks

Q3: I will wait... Thanks.

aldatsa, 2017-01-17:
eu is part of the ISO 639-1 standard. There are two regional subdivisions: eu_ES and eu_FR. I used Standard Basque in my translation so I thought the best option was to use eu without country codes. I have used eu in many projects without issues, but if that causes problems in this project you can use eu_ES.

bovirus, 2017-01-17:
@aldatsa 
Did you check with test image or my custom build when your run it on main window which is the name of your language in drop-down list?

procount, 2017-01-17:
Q1 - Actually it is the ES (Spanish) and es_XA (Asturian) languages that show the same native language text names. The eu (Basque) language shows "euskara" and changing from eu to eu_ES makes no difference.
When I select Asturian, many if not all translations are still in English. So I think the es_XA files need reviewing.

bovirus, 2017-01-17:
I will trying to rename translation_es-XA.ts (and relative icon) to translation_ast.ts and build it

procount, 2017-01-17:
I discovered that QT uses the 2 letter language codes as defined in ISO639-1.  Asturian is not listed in this set, but in the ISO 639-2 set of 3 letter language codes as 'ast' . So it is not supported and we would need a workaround for such languages.

bovirus, 2017-01-17:
@procount

Could you please provide link for 2 letter table ISO639-1?

I believe that the best way is to add in each language file a strings like "LanguageName" and use it.

In the first step we can remove Asturian language (that is not complete) but it's important to enlarge the width of drop down language selection window

procount, 2017-01-17:
Q3 answer: "Build" is already in the translation file and can be translated. But it may not be refreshed immediately in the title bar when the language is changed. Likewise for any translations done in the main list view of OSes such as [RECOMMENDED] or [INSTALLED]. These items of text will be correctly translated the next time PINN is rebooted. Not a high priority to change at the moment.

The format of the build date in the title is fixed by the compiler and is not readily changed without a great deal of effort, so I am not likely to fix this in the short term.

I added a workaround for Asturian (and a guess at what the language name should be called)

bovirus, 2017-01-17:
@procount 
Ok. Thanks.
Could you please tell me what files did you change and where can I get them?

procount, 2017-01-17:
When I'm happy with them I will post an update on github here.
(I just spotted a problem putting unicode text in the source code for Asturian)

I didn't try es-XA as opposed to es_XA. 
I think it's only the countries in the ISO639-2 table that are a problem, so I'm not too worried about a proper fix at the moment.

Language box is expanded.

Re: ISO639 - Is your google-foo letting you down? 
    https://en.wikipedia.org/wiki/List_of_ISO_639-1_codes
    http://www.loc.gov/standards/iso639-2/php/English_list.php

I put a copy of pinn-2121d.zip on sourceforge so I would be grateful if you could review it for any other issues,

softastur, 2017-01-17:
Ok, after 2 years of the first Asturian translation for NOOBS I have to say that Asturian has actually a only 639-2 ISO code "ast". I used es_XA for compatibility and because Android 4.4 (and below) didn't allow three-letter code languages. So use es_XA for Asturian is completely wrong, Asturian is a different language and not a variation of Spanish.

I should re-translate the software ASAP.

Finally, Asturian has these special characters áÁéÉiÍoÓuÚüÜñÑḷḶḥḤ

bovirus, 2017-01-17:
@softastur 
Thanks for your translation.
I propose if you are agree that you can start finalizing the Asturian translation for NOOBS (that is the first step) then I will apply the master english template that will update teh strings for PINN.
As you can see in NOOBS project site I restarted myself the project (as localization) collectiong all translation incluse yours).

You can find more info here: https://github.com/raspberrypi/noobs/issues/364

You can find language files inside this archive: https://drive.google.com/open?id=0ByL-fx0GbfuiNGRWcVpqdE5FUVE

If you are using Windows, you can use QTLinguist to translate your .ts language file

 QTLinguist 32bit standalone: https://drive.google.com/open?id=0ByL-fx0GbfuiRnJ5XzBCOVpNNjg

Thanks for your activities.

procount, 2017-01-17:
@softastur 
Is there a particular keyboard code used for Asturian? Do you have a special 'ast' style keyboard, or do most people use 'es' or 'us'?

bovirus, 2017-01-18:
@procount 

I believe that you could add for any language file two new fields

NativeLanguageName
KeyboardLayout

and use it to define for each language the name that appear in dropdown language selection list and relative keyboard layout.

procount, 2017-01-20:
I posted a question on the NOOBS github asking what format extended country codes should use - For example whether we should use es_XA, es-XA or ast for Asturian.
I have chosen 'ast' at the moment in order to display a specific Language name, as it is from the ISO639-2 list, but that doesn't really match with the format of zh_TW for Taiwan. So this might change and I'm open to further input, but I want to maintain compatibility with NOOBS too.

softastur, 2017-01-21:
@procount Asturian uses a variation of the spanish keyboard which includes the Ḷḷ and Ḥḥ characters as you can see on the majority of Linux distros.

As I said before, Asturian cannot use es_XA as its language code because that code is invented and also Asturian has not any ISO639-1 code, only the ISO639-2 "ast" code. NOOBS must be compatible with ISO639-2 standart and deprecate the old ISO639-1 which has a limited number of languages.

Finally, I was able to update the Asturian translation for NOOBS but i have some strings untranslated because i need to see their context. I'll keep you updated.

procount, 2017-01-31:
I have released PINN p2.1.3 which includes all the updated translations so far. 
If anyone has any additional translations I will endeavour to include them in the next subsequent release.

(I have used the code 'ast' for Asturian)

procount, 2017-03-01:
PINN updated to v2.2 and I found a few more text items that need translating.
If anyone has the time/inclination to add them, I will accept Pull Requests or simple attachments of the translation_XX.ts files. Thanks in advance.

procount, 2017-04-23:
PINN 2.4 is available on github for translation.

bovirus, 2017-05-20:
@softastur
@aldatsa
@cescp
@qsypoq
@qsypoq
@fonic
@sople1
@Akkiesoft
@kadamski
@ResonantWave
@Mangero
@diorray

Could you please update your translation for new PINN release (2.4.2)?
Thanks.

@proocunt

Could you please include zh_TW new language version (pull request)? Thanks.

closed 2017-05-20 by bovirus