procount/pinn #15: Language native name not showed right

bovirus, 2017-01-17:
In file

\recovery\languagedialog.cpp

There is this code to read and show language name in native language

    QString langcode = langfile.mid(12);
    langcode.chop(3);
    QLocale loc(langcode);
    /* Display languagename in English, e.g. German, French */
    /* QString languagename = QLocale::languageToString(loc.language()); */
    /* should Display languagename in native language, e.g. Deutsch, Français  */
    QString languagename = loc.nativeLanguageName();
    QString iconfilename = ":/icons/"+langcode+".png";

I have a problem with

Spanish (file language_es.ts) - code=es
Asturian (file language_es-XA.ts) - code=es-XA

Language name showed for both "Espanol de espana". Is it right?

I propose to enlarge the width of drop down list for language (the european portoguese native name is truncated on the right).
In the UI there is a lot of space unused and you can enlarge the width of language name list.

procount, 2017-01-17:
I suppose I can override the default QT native language name for certain language codes, but it is necessary to know what the replacement text should be.
I am guessing that Asturian is a regional dialect of Spain, so maybe "espanol de Astura" ?
I can resize the dialogs

bovirus, 2017-01-17:
Please post the file changes when you will resize the drop down window width, to test it. Thanks.

procount, 2017-01-19:
Changed Asturian language code to 'ast' and language name to "Asturian".
Dialogs resized. 
pinn-2121f.zip pushed as a test version in sourceforge /testing folder.

closed 2017-01-19 by procount