procount/pinn #16: PINN test build - feedback

bovirus, 2017-01-18:
@procount 

PINN test buil 2.-1.2d

Fixed
Enlargement of drop down language selection list -> OK
Patch for Asturian language native name OK-> OK

Problems found
I found two small problem about GUI.
I attached the screenshot to show it with my proposal.

1- https://drive.google.com/open?id=0ByL-fx0GbfuiUkdRVXBZdjJobVE

On the left the two strings are not aligned vertically with value on the right
The value on the right are translated (text hard-coded). Please add to language file.

2 - https://drive.google.com/open?id=0ByL-fx0Gbfuic0Y5TmgyQWpybUU
"Visualizza password" (Show password) with the control show move on the left (to near to the "Usa predefinita" button) and is not vertically aligned to the button on the right.
"Usa predefinita" (Use default) button is in correct position or it should be on the same line of OK/Cancel (OK/cancel/Use default)?

3 - https://drive.google.com/open?id=0ByL-fx0GbfuiWmZxVDFRU01JcEU
Could be possible to have OS description as language string in language file to have the description localized (I know that the OS name are own name)

procount, 2017-01-18:
Thanks for reviewing and providing feedback.
1. I will align and make translatable
2. I will make some more space for the translated text.
3. I'm not sure this is possible. All of this information is provided with the OSes and is independent of NOOBS / PINN so it is not available for translation with the rest of the text. It would require this information to be provided with the OSes and a parallel mechanism developed to load these translations in as required. Not impossible, but I don't think the OS maintainers would want to do this, and if it is not kept together, it could easily get out of date.

bovirus, 2017-01-18:
@procount 

1. OK. Thanks
2. OK. Thanks
3. It's just a proposal. Thanks.

procount, 2017-01-19:
pinn-2121f.zip posted on sourceforge in testing folder for evaluation.
Source posted on github on trans2 branch.

I have updated all the .ts files to include some additional text that will need translating for each language.
There are still some dialog alignment errors to be fixed - they look ok in QTCreator, but look different when run on the Pi.

bovirus, 2017-01-19:
@procount 

If the alignmet is about password window try just move up (laso if are OK in QTCreator) and just check on PI.
Could you please post the source changes in a test build on github?
I have QT SDK and I want to check. Thanks.

procount, 2017-01-19:
Err... I already did....
"Source posted on github on trans2 branch."
You do need to select the trans2 branch instead if master though.

bovirus, 2017-01-19:
@procount 

Thanks for all.

bovirus, 2017-01-21:
@proocunt

Could you please contact me at bovirus at gmail . com ?
I need a very short help by you and your knowledge about QT SDK and localization file.

procount, 2017-01-31:
Any future updates should now be done based on p2.1.3 on the master branch.

closed 2017-01-31 by procount