procount/pinn #21: pinn 2.1.4 - error during compilation under Linux

bovirus, 2017-02-05:
Running /BUILDME.sh under Linux 16.04.01 (x64) I got

( \
	echo "NAME=Buildroot"; \
	echo "VERSION=2015.02-git"; \
	echo "ID=buildroot"; \
	echo "VERSION_ID=2015.02-git"; \
	echo "PRETTY_NAME=\"Buildroot 2015.02-git\"" \
) >  /home/ubuntu/Desktop/pinn/buildroot/output/target/etc/os-release
>>>   Executing post-build script package/recovery/cleanup.sh
>>> host-fakeroot 1.18.4 Extracting
bzcat /home/ubuntu/Desktop/pinn/buildroot/dl/fakeroot_1.18.4.orig.tar.bz2 | tar --strip-components=1 -C /home/ubuntu/Desktop/pinn/buildroot/output/build/host-fakeroot-1.18.4  -xf -

bzcat: Compressed file ends unexpectedly;
	perhaps it is corrupted?  *Possible* reason follows.
bzcat: Success
	Input file = /home/ubuntu/Desktop/pinn/buildroot/dl/fakeroot_1.18.4.orig.tar.bz2, output file = (stdout)

It is possible that the compressed file(s) have become corrupted.
You can use the -tvv option to test integrity of such files.

You can use the `bzip2recover' program to attempt to recover
data from undamaged sections of corrupted files.

tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
package/pkg-generic.mk:103: recipe for target '/home/ubuntu/Desktop/pinn/buildroot/output/build/host-fakeroot-1.18.4/.stamp_extracted' failed
make: *** [/home/ubuntu/Desktop/pinn/buildroot/output/build/host-fakeroot-1.18.4/.stamp_extracted] Error 2

With previous pinn version in the same Linux configuration I could compile it without errors.

procount, 2017-02-06:
No idea. I suggest you force buildroot to re-download fakeroot and try building it again <shrug>

bovirus, 2017-02-07:
@procount 
Thanks.
I rebuild it from scratch and now it works.
How can delete al previous partial compilation?
What are the folders/files to dele? buildroot/output?

procount, 2017-02-07:
Yes. Have a look at the builroot manual on building specific packages and the .stamp* files.

bovirus, 2017-02-10:
OK. Thanks.
You can close this thread.

closed 2017-02-10 by procount