procount/pinn #26: PINN 2.3.0 - Italian language udates

bovirus, 2017-03-06:
@procount

Hi.

This the italian language uopdate for PINN 2.3.0 (200/200 strings translated)

[PINN-2.3.0-translation-it.zip](https://github.com/procount/pinn/files/821754/PINN-2.3.0-translation-it.zip)

procount, 2017-03-06:
Cool!, thanks.

bovirus, 2017-03-06:
@procount 

Could you please update the Italian language into the repository? Thanks.

procount, 2017-03-07:
Source repository updated.
You can test the zip file by downloading https://sourceforge.net/projects/pinn/files/archive/pinn-23a.zip/download

bovirus, 2017-03-07:
@procount 

Thanks for new build.
I propose you to change filename from "pinn-23a.zip" to "pinn-230a.zip" (to highlight that pinn-23a.zip is version 2.3.0a)
What are the differences between pinn-lite (which version is) and pinn-23a.zip?
Thanks.

procount, 2017-03-07:
pinn-lite.zip is still p2.3. I haven't updated it yet. But if you are happy with the translations, I will update it.

bovirus, 2017-03-07:
Thanks. 
For me v. v. 2.3.0a it's OK.

closed 2017-03-14 by procount