procount/pinn #27: download files only option

nutpantz, 2017-03-06:
i do a lot of stuff offline.
in  fact my pi is used mostly to download things for uses on other systems and devices
it would be nice if there was a way to download the os files to a usb drive only,  to use for later.

procount, 2017-03-06:
Me too. You can do this already manually.
Download the files in http://downloads.raspberrypi.org/<OSname> excluding the archive and images folders. Extract the marketing.tar file. Put all the files in the /os/<OSname> on your USB stick.

But automating this process is already on my roadmap :)

Once you have the OS files locally, you can then start thinking about custom installs by using the noobsconfig features of PINN. So instead of installing the plain vanilla Raspbian each time, you can get PINN to add your default wifi connection details, or workshop resources, etc.  And this customisation is kept separate to the OS, so you can update the OS and still use the same customisations (if compatible).

See https://github.com/procount/noobsconfig for details.

Thanks for the suggestion.

nutpantz, 2017-03-06:
Thanks for the info.
I usually use ubuntu mate or lubuntu 16.04.02 but they are not supported by noobs and I have not found a download for them that is not in .img format..

Do you think you will ever support .img format os files?

procount, 2017-03-06:
Ubuntu mate 16.04 LTS coming soon...
 Watch this space!

procount, 2017-03-11:
UbuntuMate 16.04.02 LTS is now available

nutpantz, 2017-03-13:
fantastic....
next stupid question....
where can a download it manually?
( not that i would ever suggest a larger workload for you or others but it might be easier for newusers to have download link for complete zip files for offline installs(but then maybe im the only one still using most of my stuff offline))

thanks for the effort

procount, 2017-03-13:
https://sourceforge.net/projects/pinn/files/os/ubuntuMate1604LTS/
(Already on my roadmap....)

nutpantz, 2017-03-13:
fantastic...
could i request lubuntu 16.04.02 also?
( still trying to get pulse audio to not stutter in mate)

procount, 2017-03-13:
link?

nutpantz, 2017-03-13:
sadly the only link they have is a torrent ( same with x ubuntu also)
https://ubuntu-pi-flavour-maker.org/download/

procount, 2017-03-13:
More like an option to download the distribution files to USB for offline installation. It may not necessarily be a zip file.

nutpantz, 2017-03-13:
I understand. Which is way originally I was hoping for a way to only download (with resuming if interrupted of course, if you got a wish, wish for the moon) the files in pinn (like maybe to usb storage(I tried pinn online and after 2 hours at .6mb/s it failed (I forget the error))
Know of any up to date ftp mirrors for the files? 
SourceForge is painfully slow for small files especially if it's not worth the time to change mirrors.(I was getting 50-100 kb/s unless I hunted a better mirror (which is only worth doing for large files as the whole site is painfully slow on the pi)
I tried googling pi ftp mirrors and after trying 3  that were dog slow ended up downloading an old version by mistake at 600 kb/s.

By which time my reasonable speed internet access time had ended.

Thanks for the effort..

procount, 2017-03-14:
I hadn't noticed Sourceforge was so slow. It does allow me to track how many downloads are made, though. Is github any faster for you?

nutpantz, 2017-03-14:
Both of them change all the time for speed.I know I can torrent up to 3mb/s, mega.nz at 1.5 mb/s, but github and source forge are rarely better than .5mb/s and some times down to .05mb/s. Ftp would of course be better then it can be set and forget until it's done. Which is why I asked about the all in one zip file. I don't mind it taking 2 hours as long as it does not fail without resume support. And even that's better than having to sit there watching it and continuously keeping it going.I've been downloading phone roms since Windows mobile 2003 and Linux almost as long, and I swear even though speed has greatly improved reliability and choice has been reduced.
(Let's not even talk about the old days when files were split into 5mb chunks with pars because the servers were so unreliable but that is getting very very old)
But I digress
I can get good speed from source forge, but I have to switch mirrors every time.some times I have to check every mirror for a reasonable speed.
but to check mirrors for the fastest speed is not a feature they have built for ease of use, and their full page advertisement can bring a fast computer to its knees let alone fastest portable one I have to use.

What ever works best for you.stick with that.
What works best for me where ever is one big file and me with my fingers crossed that it can resume.
Whatever you provide I'll work with.

procount, 2017-11-01:
Download option now supported in V2.4.3

closed 2017-11-01 by procount