procount/pinn #33: Documentation - clarify the term PINN-Lite

procount, 2017-03-08:
From 321
Edit3
https://github.com/procount/pinn/blob/m ... ME_PINN.md
Where it says

> The latest version of PINN-lite can be downloaded from sourceforge.

at the top, there is nothing to suggest what PINN-lite is, and thats what people need to download if they dont want to compile it using github.

https://github.com/procount/pinn/blob/master/README.md
Is similar to the above point, in that theres nothing to suggest what PINN-lite is, so some people might interpret PINN-lite as being like raspbian-lite, in thats its a headless os, when in fact PINN-lite is not really a headless OS.

procount, 2017-03-08:
Being derived from NOOBS, PINN-LITE is equivalent to NOOBS-LITE (i.e. no OS supplied).

As PINN supports VNC and SSH, it can actually be used headlessly (although its functionality as an OS is quite basic)

typxxi, 2017-03-13:
I had similiar trouble cause in the beginning there was also a pinn full with raspbian available if I remember right.

If there is a lite version then what / where is the full version - or did I miss something?

procount, 2017-11-01:
Terms clarified in latest documentation. (There never was a PINN full).

closed 2017-11-01 by procount