procount/pinn #37: alt_image_source Allow multiple files?

matthuisman, 2017-03-12:
In NOOBS, you can have multiple files in repo=""
Would be great if PINN would also support this to allow for repos from Pinn + Official + Other

procount, 2017-03-13:
I think it already can. I must check on syntax.

matthuisman, 2017-03-13:
Also, does PINN have a way to replace the default OS file?
Seems hard-coded in?

I have made a proxy script to allow resizing maximised partitions:
https://github.com/raspberrypi/noobs/issues/139#issuecomment-286259180

With NOOBs Lite it can work as you can set repo="" in the cmdline and it will use just what is in there.

Also, does PINN allow for a os_list file to override when in the root of the SD card?

Is there any particular reason you didn't keep the way NOOBS does repo="" and just have the two repos (default + PINN) in it when you ship PINN?

closed 2017-03-13 by matthuisman

procount, 2017-03-14:
Just try repeating the keyword e.g.
alt_image_source=http://url1 alt_image_source=http://url2 etc.

I didn't implement the feature of allowing os-list_v3.json to be on the SD card root (which was only there for testing)

The reason it is "alt_image_source" and not "repo" is because I did it first - NOOBS didn't follow me!