procount/pinn #38: Still confused about how PINN works

ilu33, 2017-03-13:
Please forgive me for kind of abusing the issue system:
After several hours of reading up on NOOBS and PINN I'm still confused about how this stuff works. A lot of people keep asking the questions I have but I can never find an answer.
I understood that NOOBS somehow uses tar archives instead of partitions. Since "PINN Is Not NOOBS" I'm hoping that PINN will work with normal ext4 partitions instead. Also I don't need PINN to install anything, to download anything or to recover anything. Ideally it shouldn't touch the manually installed OS at all. I just need a bootmanager.
Will PINN work that way? And which config files do I need to change to get PINN to load the preinstalled operating systems?

procount, 2017-03-13:
I can certainly help answering that question, although most of the information is already out there.
PINN is a fork of NOOBS with additional features, but essentially it works the same way.
You're terminology is a bit confused, which may explain why you are not finding the right answer.

Rather than me repeating stuff, why don't you have a read of the following wiki pages and come back to me with any questions you may still have:

https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained
https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained
https://github.com/raspberrypi/noobs/wiki/Developer-Information

It would also help if you could answer the following questions so I have a better understanding of what you want to do....

What OSes do you want to multi-boot?
Have you already "pre-installed" your OSes?
If so, how, where and on which partitions?

ilu33, 2017-03-13:
Thank you so much for coming back to me. To answer your questions:
OS I want to boot: Libreelec and 2 custom-made versions of raspbian (based on jessie and stretch). I have one OS preinstalled directly from the image but I can change this to any setup necessary.
Main point is: I want the OS on partitions, real ext4 partitions that I can mount, configure, backup and restore on any desktop machine (no MSWin in use here, obviously). I already read the first 2 links you posted  and now read the 3. one. The wikis all talk about setting up partitions but its completely unclear what ends up on those partitions.
> Edit: I forgot to add that PINN/NOOBS can install tar files, image files (of a partition) and partclone files (whatever they are)
This kind of answer is also confusing: I don't need PINN to **install** anything. I just need it to **handle** stuff afterwards. But I guess you mean to say that PINN can only handle the OS which it installed itself and nothing else? That means, PINN cannot handle OS on ext4 partitions?
It's definitely a problem with terminology. Like a lot of other people I'm looking for a bootmanager, but all we can find are install tools. There might be reasons for this but the reasons are never explained. Is it impossible to create a bootmanager? In old forum threads people started with boot scripts that seemed to work, but for some reason that ended up in bootberrys chroot stuff (in which you can't update the OS properly). NOOBS obviously came from a completely different direction.

procount, 2017-03-13:
I'm a bit pressed for time at the moment for a full answer, so I'll answer more later, but just to clear one thing up... PINN works on real partitions with real ext4/fat32 partition types that you can mount/unmount. This is different to Berryboot that uses a squashfs/unionfs file system combination and shares the kernel between all of the OSes it installs.
I guess the issue is that boot managers need to understand the OSes that are installed and create config files to help it (which is the info you originally asked for), so the boot manager often becomes an installer too so that it creates these files when it installs the OS. So often it is easiest to let the boot manager install the OSes. In PINN this is called a custom OS install (see the README.md files). If you have an OS pre-installed, then you can just package it up in a way that PINN understands and ask it to install it on a PINN SD card. It will also adjust cmdline.txt and etc/fstab to match the new partitions that the OS is on, otherwise it will not boot.
The other alternative (that you seem to want to do) is to somehow install your OSes manually (mount and dd?) and then try to understand the PINN config files so that you can create them yourself so PINN can multi-boot the OSes. 
The end result is the same. I would suggest that the first method is easier because it is already documented and PINN will take care of a lot of the partitioning issues for you. 
Perhaps you could explain why you prefer the manual method of doing it?
If you insist on doing that I can try and explain the process. In fact I think I posted on the forum about it some time ago (if only I could find it....)

ilu33, 2017-03-13:
I've seen your post and that's why I came here.
I understood that NOOBS puts the OS into a tar file that remains unchanged and all changes to the OS are stored on a dedicated partition for each OS. This is what I don't want. But you are saying that PINN can handle an OS installed on (two - boot and root) partitions? Which get changed all the time? Without using the tar file concept? 
My reason for manual install is that one of the 3 OS (stretch) needs to be reinstalled all the time for developing/testing purposes. But installation with PINN would be ok too as long as  the 2 others stay untouched.
Sorry to occupy your time. If I get this to work I will pay you back by contributing to documentation.

procount, 2017-03-13:
I'm not sure how you arrived at that conclusion, but your understanding of how NOOBS/PINN work is not correct. What you explain is more like how Berryboot works. (Where I state NOOBS, please also read it as PINN)

NOOBS originally came supplied with several operating system installation files, each tar'd and compressed with xz to take up the smallest amount of space possible on SD card (as this space is effectively wasted space : ) ). The user would select an OS and NOOBS would create the necessary FAT and EXT4 partitions and expand the OS files into that space. So each OS still has a FAT partition for the boot and an EXT partition for the rootfs. NOOBS also allows the OS install files to be downloaded from the internet if they are newer that the SD card versions.

So the .tar.xz files are just a means of packaging the OS installation files so they take up minimal space on the SD card and take minimum time to download over the internet. As I said above, an OS can also be packaged as raw .img.gz files, in which case NOOBS will 'dd' the files onto the partition. After installation, the tar.xz / img/gz files are no longer needed (unless you want to restore your partition back to a factory fresh installation again).

As the OS installation files became bigger, it became less practical to store them all on the SD card and still have enough space left on a 4GB (I kid you not) SD card to actually install any of them. So Raspbian became the only OS that was supplied and the others now have to be downloaded.

Now, back to your issue. 
Having installed your 3 OSes, you are free to overwrite any of the partitions however you like, using dd, tar, rsync, whatever. They are normal partitions. You can even resize them if you want. As long as the partitions remain with the same partition number, PINN will still allow you to multi-boot each one from its boot menu.

HOWEVER, note that PINN will not help you to copy a new solydXK release over the top of your existing installed version. At the moment, NOOBS/PINN will wipe the whole SD card before re-installing an OS, so they all have to be installed /refreshed together. (A feature I want to add to PINN). So you will have to do the copy of the update yourself. 

ALSO, bear in mind that when you multiboot OSes, they are actually stored in logical partitions within an extended partition. If you install solydXK from its own img file, it will (most likely) be an image of the whole SD card containing 2 primary partitions, the Master Boot Record and partition table. This is just a consequence of multi-booting requiring >4 partitions and not related to NOOBS/PINN. But it does mean that you will need to do some file manipulation if you want to copy (dd) your image off of your SD card in a format that is suitable for anyone to dd it back onto a blank SD card. It is another feature I want to add to PINN, but for now you will have to do it yourself.

Are you a developer/tester of SolydXK?
I am currently downloading the DX version so I will take a look at it. I may even convert it to PINN and add it to my list of OSes for download.....

ilu33, 2017-03-13:
> Are you a developer/tester of SolydXK?

Yes, tester, up to now only for the desktop version. I'm just starting out with the RPi. You are very welcome to spread SolydXRPi. Please report back here or on our forums if you encounter any issues. Since we are working on the transition to debian9 there might be issues with the system-adjustment scripts. (That's actually what I wanted to test.)

Your post cleared up a lot of confusion. First thing I read that is really clear. Thank you very much.
I will now backup my 2 SolydXRPi partions, start again with PINN, install several Raspbian-like OS (because atm I can't add anything later, right?), have a look at whatever it creates, resize the partitions to my liking, have a close look at the configuration files, copy back my SolydRPI partitions and try to get them up and running. This will take some time but now I know which path to follow. Thank you again for your help.

procount, 2017-03-13:
Now, that sounds like you understand the process!
The key file that PINN needs for the multi-boot menu is /settings/installed_os.json. Have a look at that AFTER you have installed the necessary OSes.

BTW, why is solydx distributed as solydxrpi_201701.img.tar.gz which is 1265736255 bytes long? the TAR makes no sense and actually makes it longer than necessary as solydxrpi_201701.img.gz would only need to be 1265734907 bytes long.

Perhaps you could raise it with the developers......?

ilu33, 2017-03-13:
Done. You are very welcome to contribute directly to https://forums.solydxk.com/viewtopic.php?f=79&t=6815 if you encounter any issues. But I can also serve as go-between :-)

procount, 2017-03-14:
A work in progress.....
https://sourceforge.net/projects/pinn/files/os/solydx/
It needs a proper icon png file (40x40) and some slideshow png pictures (387x288) ....

ilu33, 2017-03-14:
Here's one https://forums.solydxk.com/viewtopic.php?f=79&t=6809#p64685 but its too big. Can you convert it?

ilu33, 2017-03-16:
We don't have slides. I'll have to produce them as soon as my RPi is working again

procount, 2017-05-27:
SolydXK is now inlcuded in latest PINN v2.4.2a release

closed 2017-06-06 by procount