procount/pinn #39: choose wifi wlan0 or wlan1

nutpantz, 2017-03-13:
i have never had a issue with my wifi stick being reconized in linux
Realtek Semiconductor Corp. RTL8191SU 802.11n WLAN Adapter

but there was no option to chose which wifi i wanted to use
( the external works much better for me)

is it not support or is there no support to choose?

procount, 2017-03-13:
I think that's a NOOBS issue. Try posting there.
I think it was only ever expected that there be one wifi dongle connected, and if there was more than one it didn't matter which was used. Which one it chooses will depend on how the kernel identifies them. Have you tried swapping the positions of the USB ports over that they use? (not relevant of course, if your other wifi in the built in one!)

nutpantz, 2017-03-13:
ill post the issue there and see what happens thanks

procount, 2017-03-13:
Ah. That might have been a bad steer by me as I forgot NOOBS only deals with the Broadcom chipsets so they'd only be interested if you had a rpi3 or rpi0W AND had an external broadcom dongle.

procount, 2017-06-06:
You could try disabling the onboard wifi....

Create/Edit config.txt on the recovery partition and add:
`dtoverlay=pi3-disable-wifi`

closed 2017-11-01 by procount