procount/pinn #41: CEC support clarification... Or enhancement anyway.

pipini, 2017-03-13:
Hi and thank you very much for creating PiNN and so actively responding to questions.
I started using PiNN today hopefully, as a replacement to Berryboot that although has some neat features like CEC during boot selection and backing/restoring entire OS 's in one go, when it comes to update, backing/restoring or even function, unlike Noobs, many distros can become troublesome.

So PiNN will be my next big partner, in boot loading, mainly for those 2 aforementioned reasons.
Specifically for CEC I have a question though. It seems that unlike Berryboot, or the way a keyboard acts either in Berryboot or noobs or PiNN itself, the remote's cursor keys, instead of moving the selection of the OS you are going to boot, actually serve as a mouse, moving the mouse cursor bit by bit.
Is that intentional and serve a purpose, I can't really think of right now or this is an inconsistency that happens on my side, because it seems to me, that would be a lil more innovative if remote control's navigation keys worked as in Berryboot.

Thanks again for creating and maintaining this nice piece of utility.

procount, 2017-03-13:
Glad you like it!

Did you read the details on CEC support here -> https://github.com/procount/pinn/blob/master/changes.md#v211-features

Because CEC support varies so much I thought adding the mouse simulation would allow it to work with any remote. But it is not the only way to use it!

pipini, 2017-03-14:
Doh... Silly me.. Now you posted the link I am sure I saw it earlier, but possibly forgot about it because I was referring at too many documentations at the time. That will teach me to double/triple check documentations next time.
It's great you've done so far, so to please every use though.
Thanks again.

pipini, 2017-03-14:
Hmm.. I probably was in a rush and now I tried the channel buttons, it seems that my Philips TV, set does not pass through hdmi, anything else bar the cursor keys. Is there any way to redefine them somewhere?
Thanks in advance.

procount, 2017-03-14:
And therein lies the problem! 
The same results occur on a Sharp TV too, which is why I changed the 0 key to the play key.
It works fine on my Samsung, but not all TVs implement CEC the same.

By "redefine" do you mean map the cursor keys so that they move the selection item instead? That would mean that you were limited to only doing that and you wouldn't be able to select any of the toolbar buttons. At least at the moment you can do most things with the TV remote (except enter Wifi Passwords)

Such a redefinition of buttons would have to be User Specific to avoid unacceptable limitations perceived by other users. But that would take some work to make a good generic GUI.

pipini, 2017-03-14:
Since it passes play, I presume can pass stop, pause, record, forward and rewind possibly.
Yep, I was meaning map, or remap.... somewhere... in a configuration file... or sum, dunno...
But...Nah, don't bother. I can live with it. ;)

closed 2017-03-14 by pipini

procount, 2017-03-14:
When I was debugging the Sharp TV, I created a debug version of PINN which displayed all the CEC button codes. That version was pinn-123c.zip, but I have since deleted it. 
In /testing/ on sourceforge there is a pinn-123d.zip version. I'm not sure if it still contains the debug information or not. But if you have some spare time, it might be worth trying it. If it does contain the debug information, you will see keycodes appear in brackets on the title bar at the right hand side. Then you can see which buttons on your remote are supported and keycodes they generate.

pipini, 2017-03-14:
It's 213d... Right? Ok got that one and hopefully I'll find a spare SD card to image it and return with some results... Hopefully.
Thanks again for the support. :)

procount, 2017-03-14:
Yeah, hopefully it still has the debug code in. If you can't see it, after a few tv remote button presses, try ctrl-alt-f2 to get to the shell (U=root, P=raspberry) and see if there are any CEC codes in /tmp/debug. i might have left some in there. Sorry - I'm not able to test it at present.

pipini, 2017-03-15:
The version I tried does not seem to display any codes in any window title bar and neither menus or mouse pointer respond to remote presses. Also it seems that I overestimated my Linux abilities and draw you there to. I can't get out of recovery mode and see the debug.
Maybe let's call it a day... or year.
Considering that, when you get in boot menu you shouldn't need the mouse, you think could be easy to revert the cursor keys to navigate there and return in mouse pointer when in installing.
Anyway, don't really bother and thank you very much for the responses. I'll go with what PiNN is at the moment.