procount/pinn #42: Pinn Archlinux problem

symbios24, 2017-03-27:
Hello, great alternative to Noobs

There is a problem i think in pinns, i have downloaded the offline installation for Archlinux from the pinns page in sourceforge and it needs internet to continue the installation after investigating i found out that the 
"partitions.json" file in the tarball section in the end requests for example the "ArchLinuxARM-rpi-latest.tar.gz" that i already downloaded manuall from the pinn webpage can this be corrected please so no internet is required to download a file that already is included ??

Thank you

procount, 2017-03-27:
If you want to do an offline installation, then you have a little work to do to download all necessary support files and alter some.  (I'm not quite sure where you downloaded Arch from because it is not on sourceforge. But no matter)
1. Decide whether you want Arch1 (for Pizero, Model A, Model M, B+v1) or Arch2 (for B+v2? 2B, 3B)
2. Create a folder called /os/Arch1 or os/Arch2 according to your choice at step 1. This can be on the PINN SD card, or on a separate USB memory stick. Then copy all the files in either https://github.com/procount/pinn-os/tree/master/os/Arch1 or Arch2 respectively.
3. Download either http://archlinuxarm.org/os/ArchLinuxARM-rpi-latest.tar.gz or http://archlinuxarm.org/os/ArchLinuxARM-rpi-2-latest.tar.gz accordingly and put it in the same folder.
4. Extract marketing.tar in place. It should created a slides_vga folder. You can then delete marketing.tar.
5. Edit partitions.json.  The tarball line for the boot partition can be deleted altogether. The tarball line for the root partition should be changed to just:
"tarball":                      "ArchLinuxARM-rpi-latest.tar.gz"
(or the other filename if using Arch2)
6. It should now install offline (if I remembered everything correctly)

I referenced the Arch files externally because they are always changing and I don't have time to keep refreshing them.

symbios24, 2017-04-05:
Hello thanks for your answer, I have another problem with arch1 it does not show the main os photo in the left in the main installation menu of pinn, in arch2 it shows it. 
Do you happen to know what is causing this?

Thank you

procount, 2017-04-05:
Are you still installing them offline from a USB stick?
Is the icon missing BEFORE installation (so you are still selecting an OS to install - which might be an error in os.json) or AFTER installation, so an error during the boot selection (which might be related to #18)

symbios24, 2017-04-05:
No i install them from a memory card,the icon is missing before installation.i can select it but it has no icon

procount, 2017-04-05:
The same memory card that you are running PINN from?
Can you post the contents of /os/Arch1/os.json and an 'ls' of the /os/Arch1 folder?

symbios24, 2017-04-05:
yeah its the same card that includes pinn.

    "name":	    "Arch1",
    "url":          "https://archlinuxarm.org/platforms/armv6/raspberry-pi",
    "version":	    "20160122",
    "release_date": "2016-01-22",
    "kernel":	    "?.??",
    "description":  "An Arch Linux port for ARM devices RPi1",
    "username": "root",
    "password": "root",
    "feature_level": 35644412
    "supported_models": [
	"Pi Model",
	"Pi Compute Module",
  	"Pi Zero"
    ]

the folder includes this files: 
boot.tar.xz
os.json
partition_setup.sh
partitions.json
slides_vga
Arch.png
ArchLinuxARM-rpi-latest.tar.gz

procount, 2017-04-05:
Try copying Arch.png to Arch1.png and see if that fixes it.

symbios24, 2017-04-05:
i did that before it did not fix it

procount, 2017-04-05:
OK, I will look into it further. It may take me a while. Thanks for reporting it.

symbios24, 2017-04-05:
thanks a lot no problem

procount, 2017-04-05:
There was a small mistake in Arch1/os.json.
Please add a comma at the end of the line that reads: "feature_level": 35644412
I have already updated it on github, so you could just download the new version instead.

You also need to rename Arch.png to Arch1.png.  I will fix this shortly.

symbios24, 2017-04-05:
the icon is fixed,i check the box to the os to install but the install option is not available to press it,i change the partitions.json by adding the "tarball": "boot.tar.xz" back and the install in available to press but in the end when it reach 100% i get a message saying error downloading tarball and some other things under it and the installation fails it takes me after to an empty panel with no os to boot.

procount, 2017-04-06:
It looks like something has changed wrt Arch tarballs. I will look into it.

procount, 2017-04-06:
It seems to be a common problem: https://archlinuxarm.org/forum/viewtopic.php?f=64&t=11396

boderson, 2017-04-21:
hi procount,

any news?

btw. workaround mentioned in other thread, i.e to use old image works ok, however old arch image seems "a little" hard to find, so if anybody has trouble with this, here is one 2016.09 build I was able to find: http://mirror.nus.edu.sg/archlinux-arm/os/ArchLinuxARM-rpi-2-latest.tar.gz

rgds,b.

procount, 2017-04-22:
No. Nothing yet. I might have to try and work around it somehow....

procount, 2017-04-22:
The root cause has now been identified, so this will be fixed in the next version of PINN.

procount, 2017-04-23:
Fixed in PINN v2.4

closed 2017-04-23 by procount