procount/pinn #44: Disable update check?

matthuisman, 2017-04-10:
Hi,

I'm going to revert to an older version of PINN where autoboot.txt still works.
Is there anyway to disable the PINN Update check?

Also, would you possibly know the last version of PINN where autoboot works?
I suspect 2.1.4..

procount, 2017-04-10:
There's been no news from RPF about whether this Firmware bug has been fixed. 
I'll test my next development version of PINN just in case it has been fixed.
PINN v2.1.4 will be based on NOOBS v2.1, so it's a good place to start. I guess you already tried v2.2?

The update check cannot be turned off, but if you click the 'ignore' button, it will silence it until the next version is pushed, then you can click ignore again. I'll think about an option, but I think you only want this to avoid downloading a version in which autoboot is broken?

procount, 2017-04-10:
Good news - looks like it's been fixed in a recent firmware update so I'll try and include it in my next release. If you fancy beta-testing it early, then please dm me your email address, or pm me on twitter (@procount)

matthuisman, 2017-04-11:
I think i'll just keep supplying your latest version PINN.
I've put a note saying autoboot is currently broken.

People will be able to easily update PINN (thanks to your updated) and get autoboot back.

Quick question...

Your update doesn't break currently installed systems?

I've tried before replacing all the FAT files from different versions of NOOBs / PINN and it seems to break things (eg. it doesn't think any systems are installed etc)

closed 2017-04-11 by matthuisman

procount, 2017-04-11:
You'll have seen the discussion on your NOOBS open issue. I'm trying to apply the fixes that were indicated but not having much luck at the moment.
If you copy bootcode.bin from pinn2.1.4 onto the latest PINN, the autoboot.txt function seems to come back, but I don't know what incompatibilities that will bring. It probably won't work with the PiZero W, for example.
In any case autoboot.txt should come back in NOOBS/PINN 2.5

My Update mechanism shouldn't/doesn't break installed systems. If you upgrade manually, you have to be careful not to replace recovery.cmdline, otherwise it will WIPE all your SD card. Maybe that's the issue you had?