procount/pinn #48: Use of terms NOBS in PINN translation

bovirus, 2017-04-24:
@procount

Please check that in teh strings of translation teher is amention to NOOBS.

<source>Would you like NOOBS to create one for you?
WARNING: This will erase all data on your SD card</source>

<source>SD card has now been formatted ready for NOOBS installation. Please re-copy the NOOBS files onto the card and reboot</source>

<source>NOOBS v%1 - Built: %2</source>

<source>Please wait while NOOBS initialises</source>

<source>Network access is required to use NOOBS without local images. Please select your wifi network in the next screen.</source>

<source>Wired network access is required to use NOOBS without local images. Please insert a network cable into the network port.</source>

<source>Are you sure you want to reformat the drive &apos;%1&apos; for use with NOOBS? All existing data on the drive will be deleted!</source>

I believe that any NOOBS reference should be change as PINN.
Please check if necessary pelase change in amster and in the translations.

procount, 2017-04-24:
Thanks for spotting this. I have now replaced all instances of NOOBS with PINN (I hope)
I noticed that I had not updated the TS files properly, so I have pushed a new version.
Please can you check your last PR translations, I think they may need updating.
Thanks.

bovirus, 2017-04-24:
I'm afraid but in the new master language I lose many strings that I already tramslated (also if theese strings didn't contaiN PINN term). What's the matter?

Now I hsve to start again to do the same translation.

But is not the right way to redo the same job more than one time.

procount, 2017-04-24:
Sorry for that. I'm still learning with these translations.....Thanks for the PR.

bovirus, 2017-04-24:
@softastur
@aldatsa
@cescp
@qsypoq
@qsypoq
@fonic
@sople1
@Akkiesoft
@kadamski
@ResonantWave
@Mangero
@diorray
@s8321414

Please update your translation for PINN 2.4 as much as possible. Thanks.

aldatsa, 2017-04-24:
Basque translation updated: #51

bovirus, 2017-04-25:
@procount 

Could you please update basque & zh_tw translation (pull requests). Thanks.

procount, 2017-04-25:
Thanks for updated eu, it and zh translations which are now merged into v2.4b & released

closed 2017-11-01 by procount