procount/pinn #49: PINN CEC functionality stops working with hdmi_ignore_cec_init

edrikk, 2017-04-24:
Hi,
I'm running a triple boot (Plex Media Player, Kodi, RetroPie) on a Raspberry Pi 3, with bootup managed by PINN.

I had a couple of Kodi issues overnight on a couple of instances which resulted in my Samsung J series TV to be on when I woke up in the morning.

I have been able to correctly set hdmi_ignore_cec_init to 1 in the applications, however, PINN will still cause the TV to turn on.
Unfortunately, when I set hdmi_ignore_cec_init=1 in the RECOVERY partition, the CEC functionality no longer seems to work (i.e. my remote can no longer select the OS within PINN).

Note that this is with the hdmi_ignore_cec_INIT parameter set only.

Can anyone confirm this issue or provide a work around?

Thanks a lot!

procount, 2017-04-25:
So, in the case that your RPi reboots when the TV is off and PINN gets reloaded, you don't want PINN to turn on the TV via CEC, but you still want it to work when the TV is turned on manually, is that right?

edrikk, 2017-04-25:
Yes exactly.
hdmi_ignore_cec_init is supposed to behave exactly as you described it, with the purpose being to avoid TV from turning on / having the input switching while something else is being watched.

The current behavior behaves more like hdmi_ignore_cec which simply disables cec

procount, 2017-05-03:
@edrikk - I have repeated your test and can confirm this is the case.
Adding `hdmi_ignore_cec_init=1` to the RECOVERY partition's config.txt stops the TV being turned on as it should, but it also disables all CEC controls if the TV is turned on manually..
I wonder if @maxnet has any ideas? Does Berryboot behave the same? ( I use the same initialisation code as BerryBoot)

cmvac, 2017-07-12:
Berryboot had the same issue, but it got solved a couple of days ago. Take a look here: https://github.com/maxnet/berryboot/issues/97

procount, 2017-07-12:
That's not really a solution, just regressing back to an older image. You could probably do the same with PINN, but you may lose other functionality, like autoboot.txt no longer working.

cmvac, 2017-07-12:
You're right, that's not the way to go. Hope you'll find a solution!

procount, 2017-07-12:
Hoping for an update to the kernel to support CEC natively....

d1g33k, 2017-07-22:
Since this issue was linked to [mine](https://github.com/procount/pinn/issues/78), I decided to leave some related feedback.

_hdmi_ignore_cec_init=1_ has the same behaviour on my TV (Samsung E series). It prevents the TV from switching on automatically while also disabling the CEC controls on PINN. When it boots to LibreELEC, the CEC controls on the remote doesn't always initiate automatically. However, I am always able to activate the CEC controls manually by selecting the Pi from the AnyNet+ Sources List.

On the contrary, without this option, the TV switches on automatically with the Pi boot-up; the CEC controls only work sometimes with PINN and the same behaviour extends to LibreELEC i.e. if CEC controls work with PINN, then they work with LibreELEC or they don't work at all with both. However, when CEC controls don't initiate with PINN, unlike with the _hdmi_ignore_cec_init=1_ option, I am also unable to manually activate CEC from the AnyNet+ Sources list as it throws an error stating inability to connect with the AnyNet+ device.

Thus, I am compelled to use _hdmi_ignore_cec_init=1_ with my Pi and instead use the NOOBS Companion Add-On from within LibreELEC to boot to the other OS.

cmvac, 2017-07-24:
@d1g33k, do you know which is the LibreELEC partition? That would be two bird with one stone, since I also don't need it to boot to PINN first.

procount, 2017-07-24:
It depends.
Look in /settings/installed_os.json for YOUR installation

closed 2023-03-25 by procount