procount/pinn #53: Manual update?

pipini, 2017-04-27:
I switched on my pi and saw the window about a new update and the options I had.
I moved the mouse cursor with the cursor keys of my tv controller and pressed the OK button but the window disappeared without leaving behind the progression bar.
A few reboots later and some messed options in the install menu, duplicate or no entries, PiNN is back again to the old shape, that means I have the Mach build.
Is there any way to update manually or force it somehow, or I have to wait for the next installment?
Whatever the case don't get in trouble. I can wait whenever the new one comes out.

Thanks.

procount, 2017-04-27:
There were several March builds... which version do you have at the moment (top of main dialog window)?
v2.4b is only a a translation update. 
v2.4 is an alignment with NOOBS v2.4 to bring USB installation support in.
v2.3 is for PiZeroW
So if you are on v2.3, you are not really missing anything for an existing installation, and upgrading is only essential if you want to start again or install TO a usb storage device.
If you want to manually upgrade (or downgrade) you can normally just unzip all files in the appropriate archive/pinn-xxx.zip file over your existing PINN installation **WITH THE EXCEPTION OF RECOVERY.CMDLINE**. If you overwrite that file, the runinstaller option will wipe your existing OSes!
I would wait til the next release. It may not be too long. But it's odd that the selfupdate didn't work for you.

pipini, 2017-04-27:
Was the March 7th 2.3a.
Thanks. Yep. I'll wait. No hurries.

pipini, 2017-04-28:
For what it worths, debugging wise, I have 2 Rpi3's. The one where I used the usb keyboard-mouse combo worked fine. The second where I used the tv control, didn't. Actually there is an inconsistency in my remote and when I press the OK button the mouse cursor also goes a tad to the left and maybe that could be a reason the update gone wrong. But of course that's only an assumption or just a coincidence.

32Dexter, 2020-07-15:
Sorry, i have a problem. I have the old verrsion 2.8.5.4 and i would like to update at the actual version 3.3.4.2.
Unfortunately when i go to do the automatic update it fails to do it for some errors.
I was wondering how can I do the manual PINN update?
Thank you

procount, 2020-07-15:
What are the error messages?
Have a look at the troubleshooting page of the Wiki to find out how to view the error debug log.

32Dexter, 2020-07-15:
when i put "cat /tmp/debug/" error say: not a directory
if i restart PINN, i see the message: 
Error downloading distribution list from internet:
http://pinn.mjh.nz/os_list.json?libreelec_rpi2=19700&raspianlite=19700&retropie2=19700
Unfortunately, it no longer asks me to check for updates

procount, 2020-07-15:
Leave off the final '/' after debug. debug is the filename, not a folder.

The distribution list error is because you used Matt's webapge to resize your OSes, including raspbianlite, which no longer exists. Either revert your recovery.cmdline to the standard version, or redo your options on Matt's webpage.

32Dexter, 2020-07-15:
How do I restore the recovery.cmdline file to the standard version?

procount, 2020-07-15:
Go to the maintenace menu, select PINN and select Edit Config.
select the recovery.cmdline tab.

I don't have access to the default settings at the moment, but this is mine currently:

repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json forcetrigger vncshare ssh quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline loglevel=2 sdhci.debug_quirks2=4

32Dexter, 2020-07-15:
Ok, i have edit this:
http://pinn.mjh.nz/os_list.json?libreelec_rpi2=19700&raspianlite=19700&retropie2=19700
into this:
http://pinn.mjh.nz/os_list.json?libreelec_rpi2=19700&raspberrypioslite32-bit=19700&retropie3=19700
Now i have not error but PINN remain into version 2.8.5.4
If i would like to update it to version 3.3.4.2 what i must to do?

procount, 2020-07-15:
It should still notify you, unless you decided to ignore the update at a previous time.
Try going to maintenance menu, select PINN only, and select re-install.

closed 2023-03-25 by procount