procount/pinn #54: Merge?

matthuisman, 2017-05-08:
Hi @procount ,

Instead of me hosting a version of Pinn Lite pointing to my own OS.json file, I was hoping you could instead add in the OS's that I have and then I can just link direct to your Pinn Lite (saves me keeping them in sync).

The only two OS's you don't have are:
batocera & recalbox-unstable
(I'm not too worried about recalbox-unstable)

You can see my JSON file here:
http://files.matthuisman.nz/pinn.json

Batocera needs a few of the basic files hosted (no root.tar.xz though)
And recalbox-unstable doesn't need any files hosted.

At the moment, about 250 users per week are DL/ing "my version".
https://sourceforge.net/projects/noobs-extra/files/?source=navbar

What ya think?

matthuisman, 2017-05-08:
I will then also add in support / icons for all your Pinn OS's to my NOOBS Companion KODI add-on :)

procount, 2017-05-08:
Hi @matthuisman.

I have no problem linking in more OSes - the more the merrier :)
But there may be a better way than just merging the json files.
If I did so, the burden of maintenance would then fall on me, so every time you needed to adjust the json file to match a change in your OS, I would have to update it.

But did you know that you can specify multiple repositories on the command line? I think a better solution would be for you to host a json file listing only the 2 OSes you want to add: batocera and recalbox-unstable. I can add this to the list of repositories, so that there is the RPF repository for the standard OSes, , the PINN repository for the additional OSes I added, and the Matthuisman repository for yours.

Doing it that way means that your OSes get additional exposure to all the PINN users, but you maintain control of your json file to add/delete/modify the OSes you host, and also continue to host the storage for your OSes (as required). But you no longer have to keep them all in synch. 

How does that sound?

matthuisman, 2017-05-08:
@procount 

That sounds like a win / win 👍 

I have added a new PINN.json file to my bitbucket repo (so can track changes).
I have also removed the double up OS's that were already in yours.

Here is the link:
**https://bytebucket.org/matthuisman/files/raw/master/pinn.json**

If you could let me know once you update PINN to also use this file, then I will switch all my PINN D/L links to point to yours.

Also, with your automatic update mechanism - any way to move over existing users of my old PINN cmdline to the new combined cmdline?

On another note:
I also plan on making an online app to create custom PINN installs that set the partition sizes.
So, like this:
http://www.matthuisman.nz/2017/03/set-partition-sizes-in-noobs-lite-extra.html

But, it will generated from a website.

eg: 
Website fetches all available OS's.
User selects which version Pi they are using.
Available OS's shown for that Pi.
User enters the size of the SD card.

They can then set the partition sizes and choose which partition get's maximise.

Then the site simply generates the cmdline file with the correct proxy URL, adds this into the PINN files and gives them the file to download.

Do you know much about the NOOBS auto-install system (used for schools etc)?
That would be even cooler.
Then they just insert the SD card and it starts installing the pre-selected OS's.

procount, 2017-05-08:
OK, I can add your JSON file. I'll let you know when as I am in the middle of something at the moment.

Regarding update, what do your users have installed already - Is it a version of NOOBS or PINN that you have modified to use your own repo, without the self update?

I know a fair bit about NOOBS, but what do you mean by the "auto-install system used for schools etc?"
NOOBS (and PINN) already has a feature whereby if there is only 1 OS available it will be automatically installed with the 'silentinstall' option. Some other devs have added a feature to automatically install a list of OSes. I plan to do something similar in the future.

matthuisman, 2017-05-08:
Its PINN I use.

i have updated my NOOBS Companion to now detect PINN / NOOBS.
If you have a PNG logo for PINN, please let me know so I can show that for PINN (currently just using Raspberry logo)

I was hoping there was some cmdline argument or file to tell it which Os's to automatically install on boot.

That way, the generated PINN install files could contain this and automatically install their chosen OS's from the web app (stops them maybe selecting the wrong OS they haven't set a partition size for)

maybe simply an auto_install.json that just has the matching OS names or a cmdline argument?

procount, 2017-05-09:
> Also, with your automatic update mechanism - any way to move over existing users of my old PINN cmdline to the new combined cmdline?

The self-update feature replaces all PINN files EXCEPT recovery.cmdline, because that would introduce the 'runinstaller' option which would wipe the SD card when PINN restarts (not good) and would remove any other options the user has specified. However, a way of providing the URL of the repositories would be good, as manually typing it in can lead to errors. I will have a think...... 

> If you have a PNG logo for PINN, please let me know so I can show that for PINN (currently just using Raspberry logo)

No I haven't thought of a logo What size png file do you need. (Another thing for me to think about...)

> I  was hoping there was some cmdline argument or file to tell it which Os's to automatically install on boot.

Yes, this is what I was thinking of doing. It is on my list. Maybe I will raise it's priority.

> Also, I see this "Supports USB BOOT and USB ROOTFS" in your changelog. How does this work with autoboot.txt and reboot {partition_number} ?

I have not played about much with autoboot.txt on USB yet, but AIUI it should just work. Autoboot.txt boot_partition should refer to the partition on the drive that was booted. It is a firmware feature and nothing to do with NOOBS/PINN. (It is currently broken in NOOBS 2.4 and works with _some_ oses in PINN 2.4 which I modified.) You should perhaps check your OSes are autoboot compatible by having the latest firmware.

There are 4 scenarios that I can think of:
1. Using the SDcard as now with no USB device. - autoboot works as it always did.
2. PINN boots off the SD card, but Rootfs are stored on the USB, - in this case autoboot still refers to the boot partition on the SD card, whcih will then load the rootfs from USB, but it must use PARTUUIDS.
3. On RPI3 boards with USB boot enabled, PINN will boot off the USB and boot_partition will refer to the partition on the same USB.
4. PINN is on USB and SDcard on a USB enabled RPI3 - In this case the SD card will boot and the boot_partition should boot off the SD card.

> Do you know if they have plans to allow this to also work with USB partitions?

It will already but only on the RPI3 with USB boot enabled, depending on whether a bootable SD card is present, but I'm not aware of any `sudo reboot /dev/sda1` functionality.

> Maybe we can arrange some way of me telling PINN which Os to boot into on reboot?

It's already available from the boot menu - "sticky default boot", or by autoboot and I think there is a NOOBS option called default_partition. Or do you mean a cmdline line option like "boot=Retropie1" to specify a name rather than a partition?

matthuisman, 2017-05-09:
For logo, nothing huge. 
Maybe just at least 500x500px.

Yeh, for the repo's, it would maybe an idea that repos can be remotely added without cmdline needing changed.

Possibly allow PINN's cmdline to also accept a "Os list file URL"
This file contains URLS to actual os_list files.
This file could be hosted on your github.

```
{
'https://downloads.raspberrypi.org/os_list_v3.json', 
'https://bitbucket.org/matthuisman/pinn-os/raw/master/os_list_v3.json',
'https://raw.githubusercontent.com/procount/pinn-os/master/os/os_list_v3.json',
}
```

PINN fetches this list and fetches each OS file as usual.

Then you could easily add future OS contributors.
Also making PINN able to fail on a file and then just ignore it would be cool.
Or, if the JSON file also has a name for the OS Source (eg. Matt Husimans).
It could also give an error "Failed to fetch Matt Huismans Repo"
However, all the other repos should still load.

Also you could remove the hard-coded default_source from PINN and just have the raspberry pi file in this file. So, if they switch file, you can just edit github file without pushing an update.

I'll have a play around with USB booting and see how it works!

There is a optional paramater to "reboot" in Linux kernel that allows setting the partition number to reboot to. eg. reboot 5  (reboots to partition 5).
That's what my add-on uses to boot into each OS from KODI.

However, I hope it may still work with USB :)

procount, 2017-05-09:
> Possibly make PINN's cmdline to also accept a master list of repos

Funny, I was thinking along similar lines, because the cmdline was starting to get very long!

> Also making PINN able to fail on a file and then just ignore it would be cool.

If it can't load a repo file, it currently displays an error message, but continues loading the others.

> There is a optional paramater to "reboot" in Linux kernel that allows setting the partition number to reboot to. eg. reboot 5 (reboots to partition 5).

In the Pi, the partition number is written somewhere in the /proc file system, which puts it in a non-volatile register that survives a warm boot. I guess `reboot` is a higher level command that does just this but it is not supported in NOOBS/PINN itself.

matthuisman, 2017-05-09:
Yup :)

That is all good. I was just kind of "thinking out loud" in terms of if USB only OS's will work with that same reboot partition method. It's not a PINN / NOOBS issue at all.

I'm going to now install Raspbian Lite and set the USB boot bit.
Then I assume I just copy PINN to a FAT32 formatted USB stick and away I go.

I know my add-on will fail for this (as I only look for /dev/mmcblk in the installed_os json files) but I can update it to support both once I see how it works.

procount, 2017-05-10:
The problem with USB drives is that you cannot rely on the mountpoint always being /dev/sda if there is more than one. The kernel may swap them around depending on USB port or startup timing. That is why Raspbian by default now uses PARTUUIDS. So you need to handle those formats as well as /dev/xxxx

matthuisman, 2017-05-10:
So, are the PARTUUIDS in the installed_os.json file of PINN / NOOBS?

My add-on reads that file and then extracts the partition number from the partitions list for each OS in that file. Then it knows which partition number to parse to reboot to allow booting into that OS.

I guess I'll try USB boot with Raspbian to be sure.

procount, 2017-05-10:
TBH I haven't looked - it was just an indication that you should check it, But they are certainly needed for recovery.cmdline and fstab. Installed_os.json is for info only so maybe it didn't get updated. The partition number will be correct anyway, even if the drive is not. If it does put parrtuuids in there, you would just need to translate it to the device/partition.

matthuisman, 2017-05-10:
Hmm, yeh. 
I will need to investigate and try a few things out.

Oh, headups - your "main" download file on PINN is showing as xbianboot-rpi.tar.xz
https://sourceforge.net/projects/pinn/
(by default it shows latest updated file)

If you go to pinn-lite.zip file you can make it "default download" for all devices:
[image]

procount, 2017-05-10:
Thanks for spotting that. Sometimes it changes it to the last uploaded file, but mostly it doesn't, so I often forget to check and reset it back again.

matthuisman, 2017-05-11:
OK. doing some very scientific USB tests...

Consisting of 3x identical USB drives (1x with PINN + Raspbian + LibreELEC + OSMC installed)

The installed_systems.json file only contains /dev/sdaX for all systems (including Raspbian)

Switching USB drives around between ports doesn't seem to change the /dev/sda...

I am using the Pi3 USB only mode...

Possibly a different story with USB ROOT only and SD BOOT (via Pinn menu).

procount, 2017-05-16:
Uploaded for testing. 
https://sourceforge.net/projects/pinn/files/testing/pinn-241e.zip/download
The addition should be obvious - checkout recovery.cmdline.

matthuisman, 2017-05-16:
@procount 
That only seems to get the Raspberry Pi Systems for me (No PINN or Matt Huisman systems)

procount, 2017-05-16:
Hmmm. It worked locally for me but I haven't had chance to try it now I pushed it to the net. Maybe I typed something incorrectly.. will check.

matthuisman, 2017-05-16:
@procount 
A couple other small things

Should use https://  in the repo_list= in cmdline (saves a redirect)
So: https://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json

In repo_list.json, you should also use https for the PINN url.

And missing a space in "Raspberry Pi" name :)

procount, 2017-05-16:
Redirect is necessary to get https working right. ;)

procount, 2017-05-16:
Appears to be ignoring cmdline repo_list. But if you download the file manually and put it in the root partition you will see how it is supposed to work.

procount, 2017-05-16:
Fixed now. Just pushing it to sourceforge with same name so wait for it to be mirrored then you can try it. (Delete repo_list.json from sd card first though)

matthuisman, 2017-05-16:
@procount 

Awesome! That works great.

Oh, so repo_list always needs http?
However, the other URLs are OK to be https (eg. my ost_list)?

Next idea be adding (LIST NAME) next to the name of each OS?

eg.  
Raspbian (Raspberry Pi)
rtandroid (PINN)
batocera_RPI3 (Matt Huisman)

Also, how do you think it handles 2x systems using the same name?

Also, would you consider / is it possible to sort the list alphabetically?
Mainly just so systems with the same name are next to each other (recalbox-rpi3 & recalbox-rpi3-unstable)

matthuisman, 2017-05-17:
Here's a sneak peak at what I'm currently working on
**http://files.matthuisman.nz/noobs/**

Currently just lists all the OS's for chosen Installer / Board combination.
I next will make it that you can select OS's and set their Sizes.

Then it will generate a cmdline with a correct URL to parse through my PROXY script which sets the system sizes.

procount, 2017-05-17:
https needs to know the correct time, which it can get from the file server, but only if it connects via http first.

I've added name to the json file, but I'm not using it yet. I'm weighing up the best way to use it. [RECOMMENDED] and [INSTALLED] can already get added to the OS name in the dialog box, so adding (Matt Huisman) as well may well make it too long to display. I was wondering about changing the icon at the right of each line, by overlaying the number of the OS_List on top of the icon. Not as informative and I may have to add a key to relate the number back to the OS-list to be useful. Dunno. Thoughts?

If 2 OSes use the same name, only the most recent one is shown, based on their dates.

Currently, installed OSes get listed first, and the recommended OS goes to the very top. The rest is in order of discovery. I will have a think about sorting, but probably later in my roadmap as it might be easier then.

Your webpage looks useful. I have something similar on my roadmap to set partition sizes in PINN as well, but it's probably a way off yet.

I left the default RPF os_list_v3 reference in PINN just in case of a problem with repo_list. If you add '-no_default_source' to recovery.cmdline, it will ignore it.

Regarding self updates of existing systems, I can't _change_ recovery.cmdline to add the repo_list setting in (because the same recovery.cmdline file is used for new and updated installs), but I can _add_ a new file (e.g. recovery.cmdline.new) which the user can manually replace the old file with. Saves typing it incorrectly. Thoughts?

matthuisman, 2017-05-17:
@procount 
Unless you could split the list into sections with some sort of horizontal line separating them?
Then have a title of each section "Raspberry Pi OS's" "PINN OS's"  "Matt Huisman OS's"
Not sure - it's no biggy I guess if they don't know where they are sourced from..

I like the default os_list fallback :)

Update thoughts.
I think most users will only see the update dialog when first running PINN.
I don't imagine many users go into PINN SHIFT menu after initial install.
If they are, it's probably to install a new system (and therefore doing fresh install anyway).
(I assume updates are only checked for in the PINN install menu?)

For my existing users, I will just remove the old JSON file it points too.
So, if they try to do a new install - they won't get any OS's and hopefully come to the blog and update...

Or, I could keep the old JSON file but just have a single OS with description "A fresh PINN install is required"

Or, could you maybe have "forced fresh" updates?
This warns the user the update will wipe any installed systems and then over-rides the cmdline...
This warning could only show if the user actually has installed systems (not a fresh install)

I think most users who are in the PINN SHIFT menu are either Fresh or doing a re-install anyway.

matthuisman, 2017-05-25:
@procount 

The partition size setter web app is coming along nicely

http://files.matthuisman.nz/noobs/

Pretty much done.
Just need to make it show instructions on the final page (over-write cmdline with this text)

I originally used JSON to fetch the systems on the fly but this caused issues
1) Timeouts
2) Cross Origin (used YQL which wouldn't work with some systems)

So, now I am using a cron job that will update a single JSON file on my server which contains all system information and partition info.

This makes the web app and the proxy script to be pretty much instant :)

It will also save my server some resources as they proxy script no longer needs to fetch all systems either :)

procount, 2017-05-25:
Looking good!
Will hopefully have some more OSes to add to that list soon....

matthuisman, 2017-05-25:
I'll have my cron scrap the repo_list every night, so any new OS's should be added pretty quickly :)

I'll close this issue now

closed 2017-05-25 by matthuisman

matthuisman, 2017-05-27:
@procount 

I have now pretty much completed the System sizer web app :)
It now shows a count of the systems between installers (25 vs 9 at present)

I have also given it a new home at

**http://noobs.matthuisman.nz**

[image]

Also added some color and finished the instructions page :)

procount, 2017-05-27:
Yeah, I noticed when the link stopped working and. I saw your [release] post with the new one. Looks great.