procount/pinn #55: RTAndroid partitioning script error

win8linux, 2017-05-11:
I'm encountering an issue with installing RTAndroid. The install fails with this error:
`cpio: invalid option -- o`

This occurs after the following part of the partition setup script:
`cpio -o -H newc`

procount, 2017-05-11:
Could you please try it with pinn-23a.zip?
You will find it in the archive folder on sourceforge. Ignore any request to update to the latest version when it starts! ;-)

PINN 2.4 was a complete rewrite to align with NOOBS 2.4, so it is possible I missed something. Please report back if that works and if so I should be able to fix pinn 2.4 correctly.
Thanks.

win8linux, 2017-05-11:
I just tried installing RTAndroid with PINN 2.3a and it works! This issue seems to specifically affect PINN versions 2.4 onwards. Maybe something with the cpio/busybox binary?

procount, 2017-05-11:
Thanks for confirming. I will fix it in the next release

closed 2017-05-11 by win8linux

win8linux, 2017-05-12:
Is there a plan to around when the new release is uploaded or is it on a "when it's ready" kind of timeline?

procount, 2017-05-12:
I have tested the fix and it works. If you are desperate I can upload a test version. 

Right now I am working on a new feature so it will be released when it is ready.

win8linux, 2017-05-12:
Sure. that'd be nice! :+1:

procount, 2017-05-13:
I put pinn-241d.zip in the sourceforge testing folder.
If you have a chance to confirm it is now ok, I would be grateful.

win8linux, 2017-05-15:
I apologize for replying late; i had a couple of days of tending to my sick mum.

Anyhow, I've tested RTAndroid on PINN 2.4.1d successfully without issue.

procount, 2017-05-15:
No need to apologize. We all have additional constraints....
Thanks for testing.