procount/pinn #56: New os - LEDE

valdi74, 2017-05-12:
I have prepared the files needed for installation of LEDE operating system (ex OpenWrt). The files can be downloaded from:
https://github.com/valdi74/lede2rpi/releases/download/17.01.1/lede2RPi_17.01.1.tar
https://github.com/valdi74/lede2rpi/releases/download/17.01.1/lede2RPi2_17.01.1.tar
https://github.com/valdi74/lede2rpi/releases/download/17.01.1/lede2RPi3_17.01.1.tar

valdi74, 2017-05-12:
I will try to prepare some slides_vga. This is very small os (few MB per partition), so it installs very fast :-)

procount, 2017-05-12:
I noticed - the BOOT partition is larger than the ROOT partition! Maybe 1 slide will be sufficient :)

valdi74, 2017-05-12:
Slides included in marketing.tar.

procount, 2017-05-18:
Please add  `alt_image_source=http://raw.githubusercontent.com/procount/pinn-os/master/os/os_list_lede.json` to recovery.cmdline and see if it works for you.

valdi74, 2017-05-18:
I think, that for RPi3 (LEDE3) should be '"supported_hex_revisions": "2082"' instead of '"supported_hex_revisions": "1040,1041"'.

procount, 2017-05-18:
Well spotted! That's why I put it out to test first.
Should be fixed now.

valdi74, 2017-05-18:
...and replace:
https://raw.githubusercontent.com/procount/pinn-os/master/os/LEDE
with:
https://sourceforge.net/projects/pinn/files/os/lede2RPi
or upload binaries to github ;)

I don't know if file:
https://sourceforge.net/projects/pinn/files/os/lede2RPi/os_list.json
is necessary? I did not include it in my images.

procount, 2017-05-18:
I obviously changed my mind halfway through doing this. I spotted some other mistakes too, so I will fix them this evening when I can test them.

procount, 2017-05-18:
RPI3 now downloads and executes successfully. Please try it now

valdi74, 2017-05-18:
Great. I try to test it in few days.
I will modify my scripts, to match file/dir naming standard from os_list_lede.json.

valdi74, 2017-05-20:
I tried "alt_image_source=http://raw.githubusercontent.com/procount/pinn-os/master/os/os_list_lede.json" option, but download error occurred (on every file from os_list_lede.json). So I copied all the files to my private server do directory "http://valdi.piastlan.net/pinn_test" and added LEDE to os_list_v3.json and everything works fine. LEDE installs properly. It looks like there is a problem with the alt_image_source parameter (pinn-lite v2.4.2).

procount, 2017-05-22:
Hi @valdi74,

I'm afraid I kept getting in a mess with your files as they were slightly different to my others, and as I am juggling about 5 things at once and remembering all the URLS for github and sourceforge kept getting me confused. So I decided to integrate LEDE directly into my OS scripts to be the same as my other OSes, so I will take the original image from the lede website and use your supplementary files as templates for all the additional PINN meta data. This will produce a much better and consistent process. (I'm trying to process about 16 different OSes at the moment!)

But I have a question for you.
After LEDE has booted, it has mapped /boot to /dev/mmcblk0p1. This is not really correct as it should be /dev/mmcblk0pX depending on which partition PINN puts it. But Openwrt does not seem to have a /etc/fstab file where this auto-mounting is done. After googling I discovered openwrt uses /etc/config/fstab, but that doesn't exist either in LEDE. Do you know where this setting is located so that PINN can modify it during installation?

valdi74, 2017-05-22:
By default there is not fstab in LEDE/OpenWrt , because it's not needed in typical LEDE installations. To create fstab, you need to install package block-mount:
`opkg update && opkg install block-mount`
then you can insert default configuration:
`block detect > /etc/config/fstab`
and then edit it by uci interface.
Source:
[https://lede-project.org/docs/user-guide/fstab_configuration](url)
This can be done with autostart script in /etc/rc.local but i'm not sure if it's needed.
You can also manually mount volume by putting mount command in /etc/rc.local (no block-mount package needed).

procount, 2017-05-22:
Thanks. Perhaps I will leave it for now.

valdi74, 2017-05-23:
Found it :-) It's hardcoded in /lib/preinit/79_move_config file. I upgraded partition_setup.sh. New files:
[lede2RPi_17.01.1.tar](https://github.com/valdi74/lede2rpi/releases/download/17.01.1/lede2RPi_17.01.1.tar)
[lede2RPi2_17.01.1.tar](https://github.com/valdi74/lede2rpi/releases/download/17.01.1/lede2RPi2_17.01.1.tar)
[lede2RPi3_17.01.1.tar](https://github.com/valdi74/lede2rpi/releases/download/17.01.1/lede2RPi3_17.01.1.tar)
[os_list_lede.json](https://github.com/valdi74/lede2rpi/releases/download/17.01.1/os_list_lede.json)
URLs and dirs are modified to your standard (I hope so).
I try to test it in few days and let you know.

procount, 2017-05-23:
Good work!
(My standard naming keeps changing whenever I port in a new OS! So don't worry about it)
Thanks

procount, 2017-05-24:
Please try `no_default_source alt_image_source="http://raw.githubusercontent.com/procount/pinn-os/master/os/os_list_host.json` (You may also see an entry for solydxk - feel free to try that too if you like)

This includes your boot fix, a replaced RPI firmware with one that will allow autobooting, and is now totally autogenerated from source. Hopefully all ok now.

valdi74, 2017-05-27:
Now everything works fine. LEDE installs and working. Boot partition is mounted correctly. Good work! :-)

PS I tried solydxk but an error occurred during installation. Unfortunately I didn't catch the error :-| I also noticed that image displayed during solydxk installation is covered at the top by the window frame.

procount, 2017-05-27:
PINN is now updated to v2.4.2a and I included LEDE in the main distro list, along with SolydXK, AIYProjects, CStembian etc. All install nicely for me with no errors. Maybe it was a glitch or you could download the latest PINN afresh and try that?

Thanks for taking the time to create the LEDE meta files etc. 
I will close this issue now as  LEDE is now included within PINN. If you notice LEDE gets updated and you want me to include the latest vesion, just create another issue with the new URL for the main LEDE image files and I'll pull it in.
Thanks.

closed 2017-05-27 by procount

valdi74, 2018-03-25:
Unfortunately, I don't have contact with developers. I checked the forum entries from a few days ago and download the server. RPi3B+ is not supported yet, even in developer builds. We have to wait for the stable version.
PS Now the project is called OpenWrt again (after merge).

valdi74, 2018-04-28:
OpenWrt now supports RPi3B+:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=50b09196282eff24e19541003dfcb248d616006e
It is current snapshot, so may be unsatble. Now we have to wait for the stable version.

procount, 2018-04-28:
Thanks for the update.
If you can ket me know when a stable version exists I will attempt to include it.

TatharNuar, 2018-08-04:
@procount OpenWRT just released a stable version a few days ago with support for RPi3B+, v18.06.0.  Direct link to image downloads page for that model is here: https://downloads.lede-project.org/releases/18.06.0/targets/brcm2708/bcm2710/

I'm looking to put OpenWRT on mine anyway, so I can help with testing once this is added.