procount/pinn #57: Download Newer OS for OSMC and OpenELEc

tronical1, 2017-05-15:
I've been using PINN for a few months now, thank you for this tool...

I am only able to use LibreELEC with my pi0 because it is the only OS that supports the WIFI for rPi0...

I believe the newer OSMC an OpenELEC builds support rPi0 Wifi, but only the old versions of these OSes are downloaded using PINN..

Is there any way to update this?

Thanks!

procount, 2017-05-15:
As RPF dropped OpenElec in favour of LibreELEC, I just linked to the old NOOBS version of openelec. I'm not sure if OE still provide a noobs formatted version. If not I will have to create one. Possible, but it will take me a while amongst my other activities, but I'll see what I can do.

procount, 2017-05-27:
OpenELEC is now updated to the latest version.
XBian is also now included.

OSMC is pulled in from the main RPF repsoitories - you may want to ask on the OSMC forums if they will update it to the latest firmware to support the RPI0W, if they have not done so already.

closed 2017-05-27 by procount