procount/pinn #59: Adding the *BSDs as options

win8linux, 2017-05-15:
Is is possible to add the BSDs as options in PINN? The ones that are available are [FreeBSD](https://wiki.freebsd.org/FreeBSD/arm/Raspberry%20Pi) [(RPi3)](https://wiki.freebsd.org/arm64/rpi3), [RaspBSD](http://www.raspbsd.org/raspberrypi.html), and [NetBSD](https://wiki.netbsd.org/ports/evbarm/raspberry_pi/).

For users of the Raspberry Pi 3, these might help in filling up the directly interactive (disqualifying Win10 IoT) non-Linux OS spot in NOOBS/PINN vacated by the absence of RISC OS as an option.

procount, 2017-05-17:
I previously looked at the netBSD image. Fdisk (util-linux 2.20.1 on Ubunutu 14.04) only shows the one FAT partition. The rootfs partition is not identified, so I don't know how to process it.

win8linux, 2017-05-17:
The rootfs is probably using [UFS](https://en.wikipedia.org/wiki/Unix_File_System). You'd need to install additional stuff to use that FS.

procount, 2017-05-17:
That would also mean compiling UFS into the PINN kernel as well, and understanding how to adjust the partitions in UFS/xxxBSD through partition_setup.sh.  I'm not saying no, but I'm afraid it will be low priority as I have plenty of other things to do beforehand. Unless you want to provide any of these as a NOOBS image and a PR to add UFS into PINN.....

win8linux, 2017-05-17:
Does it have to be read-write? If not, then the PINN kernel could already be used as is since read-only UFS support is default since the 2.6.x versions (possibly further back).

On a somewhat related note, how does NOOBS/PINN support the comparatively more obscure RISC OS filesystem? Would it be possible to use some of the techniques used there to add UFS partition manipulation?

procount, 2017-05-17:
In a Linux rootfs, NOOBS/PINN needs to alter fstab to load the boot partition on the correct device. I have no knowledge of *bsd systems and what is required.

RISC OS uses a BLOB file containing its filesystem and RISC needs to be located at a particular offset/place on the device, which side-steps the issue. But this BLOB causes problems with Ubuntu 16 onwards as that filesystem has been enabled again and gets detected before the FAT partition table, whether RISC OS is installed or not, which is why I still use Ubuntu 14.

win8linux, 2017-05-17:
If that's what is used with RISC OS, then is it possible to make BLOB files for *BSD systems? I'm not quite sure, but I don't think that the BSDs would necessarily require specific offsets/places on the device. It is an option, though; I'd suggest placing them near the end as the last partitions to be made.

BTW I'm using Linux Mint 18 KDE (based on Ubuntu 16.04), which would explain the prescence of the ADFS kernel module in my system.