procount/pinn #63: Move OS non-tarball files to Git?

matthuisman, 2017-05-27:
Hi @procount 

Would it be possible for you to move your non OS non-tarballs to GIT?

I found that you can't hotlink Sourceforge images for starters.
Which leads to this uglieness

[image]

Also, having a git history of partition_setup / os.json / partitions.json wouldn't hurt either :)

I propose this layout:
https://bitbucket.org/matthuisman/pinn-os/src/42aee13bfeebc538d1e3f828cf70cb3a51adba46?at=master

(mimics the actual NOOBS layout).

I think you already do this for Arch OS.

Then just have same layout for the tar-balls on sourceforge.

What you think?

procount, 2017-05-27:
I didn't know you can't hotlink sourceforge files. Is it because of its redirections and mirrors? I know I had to add extra support in for sourceforge and fake a user-agent inthe request.
But it is convenient to put all the meta files for each OS in one location for maintainability. I am processing about 25 different OSes now, so it needs to be easy to deploy.

I'm not too bothered about history of the meta files, but I do like the statistics that sourceforge gives me. Arch was the first OS I added, but I don't host the tarballs, so it is a bit different and it will be updated soon to match the others.
I will have a think about the icons.....

matthuisman, 2017-05-27:
Yeh, the redirect causes the issues with the icons

Note:
I have just added KANO_OS (I saw someone request it on your PINN sourceforge)
https://bitbucket.org/matthuisman/pinn-os/raw/master/os_list_v3.json

They have a noobs zip package, so was pretty easy :)
Unfortunately they don't host the boot and root separately so I needed to extract these and upload to sourceforge.

procount, 2017-05-27:
On PINN sourceforge? I've not seen it. The only issue I had there was about OSMC....
Anyway, thanks for adding KANO.

BTW have you checked if your OSes work with Autoboot? If not, they might need a new RPI-firmware.
I have created a script that will process most distro images now and pull them apart into the right format.

matthuisman, 2017-05-28:
For the icons, I mean hotlinking (not in PINN itself).
They work fine when showing in PINN.

I could make my cache script to also download the icons and store on my server.
Then the web-app can show those.
I just wanted it as light on my server resources as possible - that's all :)
But, I guess they are pretty small...

I haven't checked the OS's.

The autoboot is quite a shambles.
Don't see why a change that broke something so badly was allowed.
Then after knowing about it, it got worst in next version - to the point of stopping booting at all.
Still no notes about this on the NOOBS page either.
And then they push the problem onto all the OS devs to update to latest firmware..

Anyway, two of my OS's are pulled direct from the devs - so I can't edit them.
KANO is the only one where I host boot and root.
I prefer to keep these completely stock to avoid possible bugs / breaks.

closed 2017-05-28 by matthuisman

matthuisman, 2017-05-29:
@procount 

I have just added RasPlex OS :)
http://www.rasplex.com/

Is there somewhere you want me to contact you when I add a system?