procount/pinn #65: CEC control - Arrows to select items (on "Select OS to Boot" window)

JeromeGH, 2017-05-31:
First, let me say thank you for this excellent project... I have also tried with noobs and berryboot, but Pinn works better for me.

Now I have 2 OS installed, "Libreelec" and "Recalbox", and I use my Sony-TV-Remote to select which one to boot.  According to [this page](https://github.com/procount/pinn/blob/master/README_PINN.md#cec-control-v211-onwards), its possible to use the arrows as mouse, and also "Channel Up/Down" buttons for "previous/next list item". Sadly, in my case this last option its not working.

My request would be to use the arrows to directly select OSs on the "Select OS to Boot" window (less clicks to select the OS), like in Berryboot ;)

Best regards

Jerome

procount, 2017-05-31:
I decided to go for the lowest common denominator which should allow most features to be used by most TVs since each manufacturer implements CEC differently.

Maybe I will allow the buttons to be programmable in the future so you can do what you want....

JeromeGH, 2017-06-05:
I can confirm now that my sony remote works fine ("Channel Up/Down" buttons for "previous/next list item")... only a little change was necessary in the tv configutarion  ;)

procount, 2017-06-05:
That's great to hear. Please could you share your configuration change for other users?

JeromeGH, 2017-06-06:
Sony Bravia 42w805a:

Settings --> Channels & Inputs --> BRAVIA Sync Setting --> Device Control Keys --> **Tunning and Menu Keys**

With this configuration now I can use the Channel Keys to control other things (Pinn, Kodi, etc) via CEC.

Best regards

Jerome

closed 2017-11-01 by procount

jcbone, 2018-10-16:
Hi!

I have the same problem, I only get mousepointer movement with my remote (by Philips) via CEC. Which is a little annoying. :-) If you could add the option to program the keys or maybe add a switch to remap the arrow keys from mouse to keyboard arrows, that would be niiiice.

Thanks, Jörg

jcbone, 2018-10-16:
Thanks! That's nice.