procount/pinn #69: New Arabic language available

bovirus, 2017-06-03:
@procount 

I added Arabic language

I modified the recovery.pro to add translation_ar.ts and translation_no.ts

Is it possible to add in recovery.pro

    translation_en.ts \

to have an empty English language file useful for new translation?

procount, 2017-06-05:
AR is is the ISO-3166-1 2 alpha code for Argentina. Please see https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2

Do you mean AE for United Arab Emirates?, 
And is this the correct flag?
[image]

bovirus, 2017-06-05:
@procount 

Yes sure. United Arab Emirates.
The flag should be O=K.

https://en.wikipedia.org/wiki/Flag_of_the_United_Arab_Emirates

procount, 2017-06-05:
Done

closed 2017-06-05 by procount