procount/pinn #73: 10-foot UI for "Select OS to Boot" window

JeromeGH, 2017-06-04:
Assuming that mouse and keyboard are, in some cases, not always connected to the Raspberry Pi, a good idea could be to redesign the "Select OS to Boot" window, and make it more friendly to use by tv-remote via CEC.
https://en.wikipedia.org/wiki/10-foot_user_interface

Maybe larger icons/texts and use the arrows to jump directly for one item to another could be a good start. Also the possibility to select a "default OS" to boot after the countdown, instead of use the last one.

Best regards

Jerome

procount, 2017-06-05:
For the moment I try to follow the UI of NOOBS. The limitations of the TV remote input are largely due to the limitations of CEC implementations. As I explained before, the choice is to allow the majority of users to use the majority of PINN features. Samsung users will have an advantage and it seems some Sony users may also find it easier, as the jumping from item to item is implemented, it just depends on the TV limitations. I may consider a programmable CEC key mapping (like in many retro game console distros) in the future so that you can change the interface, as there is no way to satisfy all users.

Regarding the "default OS" - please see https://github.com/procount/pinn/blob/master/README_PINN.md#os-boot-selector as this is already implemented

procount, 2018-05-11:
PINN v2.8.1 now includes programmable CEC buttons

procount, 2018-07-13:
Closing as the suggestions have been implemented.

closed 2018-07-13 by procount