procount/pinn #74: How are you fixing the autoboot in systems?

matthuisman, 2017-06-06:
I assume your updating the firmware to latest versions?
Are you testing quite vigorously to make sure it isn't breaking things?
What is minimum firmware required?

Would it be a better to upload latest versions of systems (not yet on NOOBS) instead where available?

eg. OpenELEC
The latest version is using 4.9 (is that newer enough)
http://openelec.tv/news/22-releases/188-stable-openelec-8-0-4-released

procount, 2017-06-06:
Don't get confused between the kernel (4.9) and the RPI firmware (i.e. BIOS) - they are different.
The linkage between firmware and the kernel is not so tight, so mostly you can change one without the other (provided there are no major changes).

I first test a distro boots normally. Then I set autoboot to that partition. If it doesn't autoboot (either freezes, or goes straight back to PINN) I change the firmware. Then I test it autoboots.

I uploaded the firmware files to https://sourceforge.net/projects/pinn/files/firmware_auto/ so you can change yours if you need to.

In PINN there is the new option of a sticky boot flag. So if Autoboot doesn't work, this can be used instead, especially if the menuboottimeout is decreased

matthuisman, 2017-06-06:
Your right, I was getting confused between firmware and kernel :)

So I just update those files in the boot partition of the OS's and then check Autoboot?

procount, 2017-06-06:
Yes, if the autoboot doesn't already work.
They must match (be from the same release) as bootcode.bin used in PINN.

closed 2017-06-07 by matthuisman

procount, 2017-06-08:
Alternatively......
If you copy the contents of my `firmware_auto` folder contents from sourceforge into a `/firmware.override` folder on your PINN SD card recovery partition, then PINN _should_ copy this firmware onto the boot partition of **any** OS it installs.

(NOTE: this is a new NOOBS feature that I have just discovered and I haven't tried it myself yet. It does copy the firmware indiscriminately, but I see no reason why it shouldn't work on most OSes)

procount, 2017-06-08:
RE: your partition resizing website, on the "Set your system sizes" page, How about:
1. A button to set all the systems to use the same partition size (subject to any minimum size contraints)
2. A button to restore to the initial recommended system sizes, after one has messed about too much with the sliders!

matthuisman, 2017-06-08:
By default, it follows the same logic as NOOBS.
Every system get's it's recommended size.
Rest of size is divided evenly among the systems.

Trying to think if this is the same as your #1.
Ummm, if we have 16000, and three systems. Evenly would be 5333 each.
If one system minimum is 6000, then 16000 - 6000 = 10,000 left = 5,000 to the others.
Left with 6000, 5000, 5000

Current method does:
Every system set to minimum let's say 6000, 2000, 3000.
This is 11,000. So, 5000 free = 1666 each
Left with 7666, 3666, 4666

So, yes - it is different.
I'll try to bring it in :)

Not sure which to have as default, or to get rid of the NOOBS method.
The NOOBS way is good as it gives the same X above minimum to everyone.

Maybe have two buttons
"Split Evenly" "Split Free Space Evenly"

Hmm, it's tricky to word - ideas?

To reset, you just go back a step and back in again.
However, a button to do this on same page is a good idea :)

matthuisman, 2017-06-12:
@procount 

OK.
I switched over the default sizes to split evenly where possible :)
And added a reset size button which takes them back to default!

Thanks for the suggestions!

I ended up switching completely over as I think this way makes more sense to the end user.

procount, 2017-06-12:
Sounds good. I must take another look. I had meant it just as an alternative for some future idea I had about managing the installed oses. Since your app makes the sizing flexible I guess either way will work.