procount/pinn #77: 2TB HDD USB Install Fail?

matthuisman, 2017-06-20:
I have a use trying to install 3x OS's to a 2TB USB HDD, but it's failing with "not enough space" when clicking install and showing a negative number.

Two images:
https://uploads.disquscdn.com/images/f6f6284c1cebad63d19015ec6efe1f03d891a52ffb9c4977dc8a4f65cf83903b.jpg?w=800&h=588
This all looks OK, Needed = 17621 and Available = 1907629

After clicking install - this error
https://uploads.disquscdn.com/images/b6520dc07cff54e9acab51a26be6632a0c4eda1467c1056e3e89d435c4a9be3b.jpg?w=800&h=588

procount, 2017-06-21:
Looks like an overflow error to me. Given that NOOBS was originally meant for installing to the SD card I don't think 2TB was ever envisaged. But now MSD is supported thise sizes are not unreasonable.

Are you able to try this with NOOBS aswell to confirm it is an underlying NOOBS issue?

Nevertheless I will look into it.

matthuisman, 2017-06-21:
It is a user from my blog. I will give him that version to try and report back :)

closed 2017-06-21 by matthuisman