procount/pinn #78: HDMI-CEC stops working on TV unless remote is used with PINN

d1g33k, 2017-07-08:
I am triple booting Raspbian/LibreElec/Batocera with LibreElec as the default boot option on an old Samsung Smart TV (I think it is 40ES5600) with AnyNet+. Under normal circumstances, although LibreElec successfully loads the HDMI-CEC plugin, there is no response from the remote. However, if I use the remote on the PINN recovery/boot selection screen, then the remote works properly within LibreElec. However, the remote works fine every time on LibreElec with NOOBS. 

Similarly, auto switching when connecting to Chromecast stops working unless I use the remote with PINN at startup. It seems that PINN disables HDMI-CEC on my TV unless I use the remote with it at startup. I would go back to NOOBS but for the fact that I want to use Batocera.

Request some means to troubleshoot and resolve this issue.

d1g33k, 2017-07-09:
A few clarifications after further testing. HDMI-CEC as a whole isn't disabled on the TV. Chromecast auto switching is working for me now (fresh installation with PINN 2.4.2) but it didn't with my previous installation which used PINN 2.4.0. Not sure if it was a Chromecast issue or a PINN one.

However, I can confirm that the HDMI-CEC controls don't work with LibreELEC when using PINN if I enable the sticky default boot option. However, I just disabled the sticky boot option and allowed the timer to run down on the OS selection screen and the remote worked fine, just as it does when I use the remote to enter recovery.

From this experience, it seems that HDMI-CEC gets disabled on the Pi when using the sticky default boot option in PINN but I wouldn't know better.

procount, 2017-07-10:
All very odd, and tricky to troubleshoot because the CEC code is all closed source so I'm not sure what is going on there. Both PINN and the selected OS will initialise CEC for their own use and in between the Pi is rebooted, so I don't understand why one affects the other. Thanks for pointing this out. I'll take a look, but it will be difficult.

d1g33k, 2017-07-11:
Thanks, but I think you shouldn't bother about it too much because the behaviour is far too unpredictable (at least with my TV). You can throw all the scenarios I spoke of previously out of the window as it seems to be mainly empirical.

Earlier today, over the course of half a dozen cold and warm boots, I couldn't get the remote to respond at all, either on the recovery screen or the boot selection one and consequently on the LibreELEC screen.

Sorry for asking a fundamental question, but is there any way to completely disable HDMI-CEC on PINN and make it the same as NOOBS post-installation. I don't want to go through the process of setting up the Pi once again.

procount, 2017-07-11:
Please see #49 for a way to disable it in PINN, although it should be hdmi_ignore_cec=1
If this is not sufficient, I will look at a PINN option to specifically disable it.

d1g33k, 2017-07-22:
I finally got around to trying out your suggestion today. Over this period, I had the CEC controls work occasionally but more often than not it failed to respond. Earlier today, I encountered the situation once again where CEC controls would't respond even after 5-6 reboots and hence it was a good opportunity to try out the config edit.

However, I think _hdmi_ignore_cec=1_ altogether disables CEC which is not what I want since I need the controls to work with LibreELEC. Hence, I went ahead with _hdmi_ignore_cec_init=1_ instead. Since then the CEC controls have worked fine with LibreELEC for a couple of reboots, so I am hoping this is the solution I was looking for. While a trifle inconvenient, I have set the sticky boot to LibreELEC and use the NOOBS Companion Add-On from Matt to switch OS. 

For those in the same situation, you need to add hdmi_ignore_cec_init=1 to the config.txt in the _root_ partition of the SD Card and also copy it to the _Recovery_ partition.

closed 2017-07-22 by d1g33k