procount/pinn #79: How to turn off the rainbow pixel?

cmvac, 2017-07-10:
How can we turn off the rainbow pixel on starting up PINN? I've added:

disable_splash=1

to the config.txt available when booting PINN, but I guess it only accesses those files for the bootable systems installed and not for PINN itself.

procount, 2017-07-10:
Will take a look....

procount, 2017-07-10:
Well, it works for me. 
I created a config.txt file on PINN's partition and added disable_splash=1 and the rainbow screen did not appear.

cmvac, 2017-07-11:
Nice! Let me try the same. Maybe the problem with the CEC connection can be solved the same way, by adding "hdmi_safe=1" to the config.txt file. It did the trick for me before, when I had CEC connectivity issues with OpenELEC. I'll try it as well and see if it works.

cmvac, 2017-07-11:
Is that the flash (FAT32) partition accessible on the SD card or another ext4 partition? I have a config.txt file on the FAT32 boot partition, but changing those values doesn't change a thing.

procount, 2017-07-11:
It's on the FAT32 partition where PINN is stored. Check your typing.
it is a function of the firmware.

cmvac, 2017-07-11:
Yes it worked. I know it is a function in the firmware, but I was trying to get to it without having to remove the mSD card from the Pi, through a modded samba.conf file. It worked by creating the file directly on the mSD. Thanks!

BTW, is there any way I can access the Fat32 partition using the samba share on Libreelec? What is the path I should add?

procount, 2017-07-12:
PINN is stored on the first partition of the SD card, formatted as FAT32.
This partition is referenced as /dev/mmcblk0p1.
To access it from another OS you first need to mount it.
For example:

``` 
sudo mkdir /tmp/pinn
sudo mount /dev/mmcblk0p1 /tmp/pinn
```

Then you need to share /tmp/pinn as a samba share.

You can also access the pinn FAT32 partition whilst PINN is executing remotely using vnc or ssh, or locally using a keyboard and accessing the recovery shell using ctrl-alt-f2. See the PINN readme.md file.
Whilst PINN is running, this partition is already mounted as /mnt (normally read-only) so you can remount it read-write using:

`mount -o remount,rw /mnt` 

and then access it via /mnt

cmvac, 2017-07-12:
That's it! I've added 

mkdir /tmp/pinn
mount /dev/mmcblk0p1 /tmp/pinn

to autostart.sh on LibreELEC. Then on samba.conf I've added a link to that folder as follows:

 [PINN]
  path = /tmp/pinn
  available = yes
  browsable = yes
  public = yes
  writable = yes

And it works! Also got inspired and mounted the Recalbox partition on the smb share. Thanks for your help!

closed 2017-07-12 by cmvac