procount/pinn #80: repository list - RIscOS RC15 - Pi 3 support

arober11, 2017-07-11:
The RiscOS RC15 release in April added support for the Pi 3, can you please tweak the repository list to enable an install a Pi 3?

FYI: https://www.riscosopen.org/content/downloads/raspberry-pi

procount, 2017-07-12:
PINN just uses the same RiscOS version referenced by NOOBS on the RPF website, which is still an older one.
The link you provided is for a raw image. This needs to be converted to be used by NOOBS/PINN, so it is not just as simple as tweaking the repository list.
Perhaps you could ask the RiscOS maintainers if they will be creating a NOOBS version, and/or the NOOBS team to see if they will use the newer version?

closed 2018-08-06 by procount

procount, 2018-10-24:
A new version of RISC_OS has now been produced and hosted by the RISC maintainers and is available for immediate installation via PINN.