procount/pinn #83: If I need to update Korean l10n file from NOOBS to Here?

sople1, 2017-07-31:
refer: https://github.com/raspberrypi/noobs/pull/428 | https://github.com/raspberrypi/noobs/pull/409

(Sorry for my English)

Hi, I'm a first author of Korean l10n file. and, I just update to noobs before few day but your project using another author's file that edited from my old file.
(I didn't know until before few day that this project existed because I didn't know until before few day that i didn't think that I receive notification to me in GitHub Notification)

So, if your project want to update to current noobs l10n file, I will update soon.
But if not, I will not touch your project.

How do I do?

procount, 2017-07-31:
Hi,
You are very welcome to update the Korean translation for PINN.
Just issue a Pull Request with any changes necessary.
Thanks.

sople1, 2017-07-31:
Oh, thanks for your "welcome" :D 
I will request for update soon :)

closed 2017-07-31 by sople1