procount/pinn #84: creating a blind-accessible, multi-OS Micro SD card

RichMorin, 2017-08-17:
I want to create a Micro SD card containing several bootable partitions for operating systems, plus a partition for common files (e.g., code, data, docs). Any running system should be able to mount any of the other partitions, allowing ease of inspection and modification.

All of the administrative functions need to be accessible to blind users and developers.  For example, the user should be able to perform any PINN task via Wi-Fi from an SSH session on some other machine.  And a pony.

**Example**

We should be able to divide a 128 GB Micro SD card into four bootable partitions (16 GB each) and one common partition (64 GB).  This will let us install substantial copies of Arch Linux, Raspbian Lite, and Raspberry VI, while leaving another partition free for some other OS.

procount, 2017-08-18:
> I want to create a Micro SD card containing several bootable partitions for operating systems, plus a partition for common files (e.g., code, data, docs). Any running system should be able to mount any of the other partitions, allowing ease of inspection and modification.

Totally possible.
 
> **All of the administrative functions** need to be accessible to blind users and developers.  For example, the user should be able to perform any PINN task via Wi-Fi from an SSH session on some other machine.  

Currently, **NO** PINN tasks can be performed via SSH, but they can be performed over VNC by sharing the screen to another computer. So we would need to look closely at what you mean by "**All administrative functions**".

Is the blind user going to:
1. prepare the initial SD card card by installing the OSes
2. be a maintainer of the OSes on the SD card
3. develop applications using the OSes on the card
4. be a user of the many OSes or applications on the SDcard

So which of the above use cases shall they need to perform? 
Or are there different blind users that will be using each use case? 
Are there specific PINN functions that they need to perform?

> And a pony.

Seriously? :wink: 

> **Examples**
> 
> For our initial development, we'd like to divide a 128 GB Micro SD card into four bootable partitions (16 GB each) and one common partition (64 GB).  

The default is to allocate the minimum SD card space to each OS, then add/allocate any remaining space equally to those partitions. But it can be done exactly as you have specified by tweaking some of the OS meta data files.

> This would let us install substantial copies of Arch Linux, Raspbian Lite, and Raspberry VI, while leaving another partition free for some other OS. 

When NOOBS/PINN is used to install the OSes, it partitions and formats the SD card, then installs all the OSes in one go. There is no current provision to add another OS to an existing card without removing all of the other partitions. It **_could_** be done if space for it was reserved at the beginning, but it would be a manual operation that requires more advanced linux skills.

> For a production version, we might want to divide a 64 GB Micro SD card into two bootable partitions (16 GB each) and one common partition (32 GB).

OK. This just means different meta file tweaks,
 
> **Motivation**
> 
> Our UI designer and developer, Amanda Lacy, is congenitally blind. So, she finds manipulating SD cards to be both difficult and risky. 

Which model(s) of the RPi are you targetting this at? If using the RPi3, have you considered the USB boot option so that she only has to manipulate USB memory sticks rather than the tiny  uSD cards?

> Connecting to a "remote" machine over SSH, on the other hand, is trivial for her (as a recent CS graduate). 

Does she use use any accessibility features of her local computer to do this (e.g. text to speech)?
Is VNC acceptable (e.g. with a screen reader) or are graphical type screens totally out of the question?

> Having an SD image that will let her manipulate and swap among OS variants should let her proceed in a convenient and efficient manner.

RichMorin, 2017-08-18:
> Currently, NO PINN tasks can be performed via SSH

Ouch

> but they can be performed over VNC by sharing the screen to another computer.

VNC transmits a graphical (ie, raster) image of the screen. A blind user can't make use of this and a screen reader (on the receiving computer) has no way to read the text to the user.

FWIW, remote access isn't the real issue here; we just want Amanda to be able to do as much as possible on her own. On a full Linux system, she might do this via a screen reader and/or a braille display. PINN probably won't support these any time soon, but having a text-based UI seems like a plausible feature to request... 

 > So we would need to look closely at what you mean by "All administrative functions". ...

I expect to "prepare the initial SD card by installing the OSes" before shipping the prototype to her. I can also prepare and send other SD cards as needed, but this introduces assorted complications (eg, manipulating cards).

The line between maintaining OSes and developing applications can be fuzzy. For example, Amanda will be working on areas such as support of the prototype's tactile buttons (eg, Perkins-style braille input, control functions, Morse code). She will also be using apt-get (or the Arch equivalent) to add packages.

Down the road, many of our expected users will be content to simply use the applications we provide. However, user empowerment is a major goal of this project, so a blind user should be able to adapt and extend the system if they're willing to put in the effort.

For the moment, let's assume that the initial SD card will contain several fixed partitions, some of which are loaded with OSes. Using SSH, Amanda should be able to log into one of these OSes and access (eg, inspect, modify) its files. She should also be able to access files on another partition.

So, for example, she could edit configuration files in the other partition and then reboot into it. She should also have a way to load a partition with an OS image, as this would let her try out new or modified OSes.

> The default is to allocate the minimum SD card space to each OS, then add/allocate any remaining space equally to those partitions. But it can be done exactly as you have specified by tweaking some of the OS meta data files.

Cool.  By way of explanation, the extra room will be needed for packages that Amanda installs, software that she creates, etc.

> When NOOBS/PINN is used to install the OSes, it partitions and formats the SD card, then installs all the OSes in one go. ...

Neither Amanda nor I have "advanced linux skills", but we can certainly follow specific instructions. Also, while it would be great to let Amanda add or change OS images as desired, we can live without this for the moment.

> Which model(s) of the RPi are you targeting this at? If using the RPi3, have you considered the USB boot option so that she only has to manipulate USB memory sticks rather than the tiny uSD cards?

The initial prototype is based on the Zero W, but we expect to support the 3B down the road. Could the USB boot option be used with the Zero W (eg, by means of a USB hub)?

> Does she use any accessibility features of her local computer to do this (e.g. text to speech)?

She uses a screen reader with a QUERTY keyboard and a braille display. Our plan is to support all of these on the RasPi eventually, but we can work our way over to this...

> Is VNC acceptable (e.g. with a screen reader) or are graphical type screens totally out of the question?

As noted above, raster images are useless to a blind user or a screen reader. ASCII text is really the most accessible format.

procount, 2017-08-18:
So, I take it that Amanda can find her way around a Linux command line shell: she can type commands, edit files, mount/unmount partitions, untar files, use dd and so on, as long as it is a textual interface?

If you will prepare the initial SD cards with the OSes preloaded, then all she needs to be able to do is:
1) Select one of the OSes to use
2) Switch from one OS to another
The other functions you mention seem to be just using a standard Linux OS. 
PINN is just used to allow the multi-booting and switching functionality. We can even make PINN disappear.

Further down the line, she may need to be able to 
3) add an OS or replace an OS in one of the partitions with a new one.

The last option will need some work to do it textually (and it requires some other enhancements that are already in my roadmap), but doable.

RichMorin, 2017-08-19:
> So, I take it that Amanda can find her way around a Linux command line shell: ...

Yep. Indeed, she is quite enthusiastic about the fact that pretty much everything in Linux can be done from the command line. Unfortunately, a lot of boot code (etc) does not have support for screen readers, braille displays, etc. If we can resolve a few remaining issues in this area (such as adding or replacing OS images), she should be able to do everything she needs to.

Meanwhile, how should we get started on preparing the initial SD cards? My preference would be to create a detailed HowTo, based on your input. If you'll send me a rough draft, I can try it out, wordsmith it, and post it to a public wiki page.

procount, 2017-08-19:
Have a read of the Readme files on this github.

The getting started section should be enough for you to start creating an SD card with Arch and Raspbian lite and have a play with it.

I will create an image for Raspberry VI for you to install. Remember you will have to start again once I have that available.

The Readme_pinn file will detail all the other things you may want to do with pinn.

There is a wiki on my github page if you want to develop some further or more specific instructions.

Meanwhile I will think about a cmdline interface over ssh.

I imagine something like a numbered menu system that can be screen read?

Any suggestions welcome

procount, 2017-08-20:
I've added Raspbian-VI to the list of OSes that are installable by PINN, so now you can install Arch, Raspbian-Lite and Raspian-VI onto the same multi-boot SD card.

procount, 2017-08-21:
I've had an initial think about how I can enable a textual interface to PINN.

There are already 2 user interfaces to PINN:
1) Standard keyboard and mouse with graphical display
2) Remote VNC interface with graphical display
Currently these interfaces work on the same graphical display, i.e. a remote user can see what a local user is doing and vice-versa. They are different views of the same graphical display.

My first choice concerns whether the textual display view is:
1) Another (third) view of the same graphical display (but described by text for screen readers etc)
2) A completely separate display that works independently of the graphical display.

Option 1 means that the textual interface will follow all graphical dialogs and pathways through the code. It will work exactly the same as now, but the menu options and lists will be described in text.

Option 2 means the interface can be completely different to the current user interface, with reduced options for only the required functions that are necessary. There may be a conflict here if the blind user were operating PINN whilst another user was using the standard or VNC interfaces and they try to do different things, but this is probably an unlikely scenario.

In both cases, my vision of the interface is menu driven like that found in the parted or fdisk tools; a bit like an 80's style Text Adventure game:
"You are at the main menu. Options are:
1) install
2) list operating systems...." etc.
With perhaps verbose and succinct modes to control the amount of text displayed for beginners and advanced users.

My second choice concerns how the textual interface is presented. Again, I thought of 2 possibilities:
1) The display is like a normal terminal session over SSH - the output from PINN is interleaved with the user input. The 2 streams are synchronous i.e. when the user is expected to type some input, PINN will wait and not be able to display any informative messages until the user has finished entering.
2) The PINN output is separate from the user input, something like it is in a chat application with a separate window for the input and output. This allows the PINN output window to be asynchronous and display messages whilst the user is typing an input. I'm not sure how this would work with a screen reader. Maybe it means opening 2 SSh sessions to PINN, one for input and one for output. It may be too complicated, but just a thought.

>The initial prototype is based on the Zero W, but we expect to support the 3B down the road. Could the USB boot option be used with the Zero W (eg, by means of a USB hub)?

No, only the RPis with the BCM2837 processor (3B and 2Bv1.2) support USB boot.

Is Amanda familiar with any text editors like VI or nano?

Have you downloaded pinn and tried using it yet?
Any feedback on that or the above is welcome.

RichMorin, 2017-08-22:
Thanks for adding Raspberry VI to the PINN options! Sadly, I won't be able to try any of this out until early September, as I'll be out of town. However, I'm definitely planning to leap on it as soon as I get back and settled in.

As a sighted person with NO screen reader experience, I'm definitely not the right person to answer questions on UI design, so I've asked the folks on the Raspberry to jump in with comments. That said, my own preference would be a command-line interface (rather than a menu), as this could be scripted.

Amanda has been climbing the learning curve on Emacs and Emacspeak. I don't know if she has had a chance to play with vi or nano, but I'm certain that she could learn enough about them to get by.

procount, 2017-08-22:
Ok no problem.
If you could provide some examples of what you mean by a cmdline interface c.f. a menu driven interface that would be great.

RichMorin, 2017-09-06:
Hmmm.  I'm not all that familiar with PINN's capabilities, let alone the actions that a text-based UI would need to support. Might you be able to post a summary of the actions, so we can categorize them, etc? We could then develop and compare menu-based and command-line versions...

procount, 2017-09-10:
Thinking about it, there's probably not a lot to do really, at least initially. As I stated above, it is probably only necessary to select which OSes to install in the first place, and then to select which OS to use on subsequent reboots.  Most of the other features you'll probably not need, but we can add as necessary. A lot will also depend on how best I can implement it. 

I'm busy tidying up some other PINN improvements at the moment, but I'll start to look at how to implement the interface and probably throw some betas at you to look at.

procount, 2017-11-01:
@RichMorin - Are you still interested in this? Did you get any feedback about how a text based UI should work?  I have been rather busy developing v2.4.3 of PINN but I am now finally in the release stages of that. I decided to depart from the NOOBS documentation and have completely overhauled the documentation structure so hopefully it flows better now.

closed 2018-08-06 by procount