procount/pinn #88: Using Wi-Fi with PINN on a RasPi Zero W?

RichMorin, 2017-09-05:
The [README_PINN](https://github.com/procount/pinn/blob/master/README_PINN.md) page says:

> If you have the official Raspberry Pi USB wifi Dongle, or are using the Raspberry Pi 3 Model B with built-in wifi, the wifi icon on the PINN toolbar will be available.

Does (could) PINN support the built-in Wi-FI on a RasPi Zero W?

procount, 2017-09-05:
Yes. 
NOOBS _only_ supports the Broadcom based wifi dongles (as you might expect).
PINN supports all that NOOBS does, plus several of the most popular wifi dongles. These should be listed in the README_PINN.md doc. The Rpi0W uses the same wifi chip as the Pi3.
I probably need to update the list to include the recent RPi0W as well.

procount, 2018-05-09:
Pi0W included

closed 2018-05-09 by procount