procount/pinn #90: need help setting up a RasPi Zero W with Raspberry VI

RichMorin, 2017-09-05:
I need help setting up a RasPi Zero W (ZW) with Raspberry VI (RVI). At this point, I have:

- formatted a 16 GB Micro SD card, using OSX Disk Utility
- downloaded PINN and unzipped it into the card
- downloaded, checked, and unzipped the RVI archive
- copied the RVI img file into the SD card's os/ directory
- wired up the ZW with:
  - a USB hub, keyboard, mouse, and power source
  - a DVI monitor, via an HDMI adaptor cable

When I power up the ZW, it goes through assorted gyrations, then brings up a "no network access" dialog: "Network access is required to use PINN without local images. Please select your wifi network in the next screen."  I have an img file in the os/directory, so I shouldn't be getting this message.

Also, the cursor doesn't respond to the USB mouse.  I saw some USB nastygrams flash by during the boot process; these are probably related to this problem, but they're hard to capture.

- "... device descriptor read/64, error -71"
- "... unable to enumerate USB devices"

procount, 2017-09-05:
PINN does not (yet) install IMG files that are designed to be written to the whole SD card. The OS needs to be formatted in a particular way with additional meta data files to tell PINN what to do with it, since PINN needs to reposition the OS to different partitions according to what else is being installed.

It is better to connect to your wifi network and install the RVI image that I have already prepared.

If you want to do it locally, then you would need to download all the files and sub folders from  https://sourceforge.net/projects/pinn/files/host/raspberry-vi/ and put them in the /os /raspberry-vi/ folder.

RichMorin, 2017-09-06:
> PINN does not (yet) install IMG files that are designed to be written to the whole SD card. ...

Pretty much what I suspected...

> It is better to connect to your wifi network and install the RVI image that I have already prepared.

As reported in Issue #88, README_PINN doesn't indicate whether the Zero W's internal Wi-Fi is supported, whether it can work with a USB-based Wi-Fi dongle, etc. Details, please?

> If you want to do it locally, then you would need to download all ...

That sounds both tedious and error prone.

> Sometimes it is necessary to add an OTG cable from the ZeroW to the USB HUB ...

The RasPi responds to the keyboard (eg, Enter). The mouse I'm using (a Kensington Optical Elite) doesn't seem like a likely culprit, but I'll try another if the problem persists.

procount, 2017-09-07:
>Details, please?

Most popular USB wifi dongles are supported. ALL the RPi builtin Broadcom and official RPF dongles are supported, plus the following list (from README_PINN.md):
zd1211rw
rtl8192cu
rt2500usb
rt73usb
rt2800usb
mt7601u
rtl8150
r8152
asix
ax88179_178a
dm9601
smsc95xx
rtl8188eu - for hubpiwi and others from v2.1
IoTpHAT
If the Wifi icon is not greyed out in PINN, then your dongle is supported.

>That sounds both tedious and error prone,

You could also download them to a USB stick instead and install from there.

> I was using an unpowered USB hub

A powered hub is normally required for the Pi0W.

> but the mouse still fails to respond

NOOBS/PINN support most keyboard and mouse combinations. Does the mouse work in normal Raspbian? Could you report the output of lsusb for the mouse? (Try running this before and after plugging the mouse in to isolate the details for the mouse only).

RichMorin, 2017-09-07:
It turns out that the problem was an incompatible keyboard. After I swapped this out, I was able to create, boot, and configure a Raspbian Stretch Lite distro, including SSH over Wi-Fi. I then tried to boot the PINN/RSL card. The Good News is that both the keyboard and mouse work.  The Bad News is that the "Wifi network selection" hangs with a cycling "Connecting" dialog.

procount, 2017-09-07:
The wifi connection bit is straight from NOOBS. No idea how to diagnose that really. I never had a problem with it.

procount, 2017-11-01:
> If you want to do it locally, then you would need to download all the files and sub folders from https://sourceforge.net/projects/pinn/files/host/raspberry-vi/ and put them in the /os /raspberry-vi/ folder.

P2.4.3 supports downloading the OS distribution files to a USB stick. You can install them from the USB stick, or copy them to the OS folder of the SD card and install them from there.

closed 2018-05-09 by procount