procount/pinn #91: Boot window not centered on the Rpi 7" Touchscreen Display

phmio, 2017-09-11:
Hi, 
On the Rpi 7" Touchscreen Display the OS window list is not centered (see picture).
And is it possible to hide the cursor when using PINN with the Touchscreen Display ?

[image]

procount, 2017-09-11:
Yes I am aware of that. It will be fixed in the next version.
Not sure about the cursor. The touchscreen can still be used with a mouse. I often find it useful to see where I last touched when I'm not being very accurate so I can tell how to compensate.

phmio, 2017-09-12:
Ok, but about the cursor, maybe you could add a flag to the argument list in the recovery.cmdline to enable/disable it...

As the back light of the touchscreen is too hight i wanted to reduce it. So i enable ssh and notice that /sys/class/backlight/rpi_backlight/brightness is missing  :(
In fact there is no /sys/class/backlight at all. Have you an idea, why?

I want to use the script /background.sh to set brightness at boot but it seems that the script does not work. The log /tmp/out.log is not created:
`echo 'hello' > /tmp/out.log`
`echo 50 > /sys/class/backlight/rpi_backlight/brightness`

I have created background.sh on the boot partion (RECOVERY) is it the right place?....maybe i missed something?

procount, 2017-09-12:
Well, I'll take a look.

It seems the backlight driver is not included, so I would need to add it.

maybe add 
`#!/bin/sh`
as the first line of backgound.sh

phmio, 2017-09-14:
Ok thanks, i will test it as soon as the new release is out ;)

I have a look at the boot init code, and seen that /mnt/backgound.sh never runs because /dev/mmcblk0p1 is not mounted to /mnt . 

I think you have to add something like this in init file:
`mount -t vfat -o ro /dev/mmcblk0p1 /mnt`
`   if [ -e /mnt/background.sh ]; then /mnt/background.sh & fi;`
`# to not disturb recovery execution unmount it`
`umount /mnt`

 In fact /dev/mmcblk0p1 is mounted to /mnt only by the /usr/bin/recovery program.

procount, 2017-09-14:
Ah yes, you're right. I guess not many people are using this feature!

procount, 2017-09-14:
Hmmm. Not quite as simple because PINN could now be running from USB on a RPi3.

phmio, 2017-09-17:
Or maybe background.sh execution could be added in the recovery program...
I have no skill on Qt, but as i understand after the call of  `drive = findRecoveryDrive()` drive seems to point to the real boot device (mmcblk0 for SD, or sda for USB) and so with `partdev(drive, 1)` we could access the right RECOVERY partition.

procount, 2017-09-17:
Yeah, It's on my list of things to do.

phmio, 2017-10-05:
Hi,
I could not use your DSI flag solution to auto-switch HDMI/DSI screen cause some TVs doesn't deassert hotplug when switched off, mine is in this case as described [here](https://github.com/raspberrypi/firmware/issues/855#issuecomment-323753200). 

So i have to play with PINN multiboot on the 7" touchscreen, but when i start my app for TV, the 7" touchscreen remains with kind of white lines quite ugly. So i have to power it off just before booting the selected TV app.

That's why i have added `runCustomScript(...)` function to PINN to run a custom script (eg your `background.sh` at startup or new one `before-reboot.sh` just before soft reboot) to add some stuff like setting brightness or power off display & co.
If you are interested by the code or want to have a look if it sounds ok,  how could i submit you the recovery (main.cpp) modifications ?

procount, 2017-10-05:
The best way is to submit a pull request.

phmio, 2017-10-15:
hi,
done :wink:

procount, 2017-11-01:
Implemented in p2.4.3

closed 2017-11-01 by procount