procount/pinn #94: Error with headless install using preconfigured wpa_supplicant.conf

hurlenko, 2017-10-20:
I don't have a monitor to connect RPi to so for headless installation I have 2 options: using Ethernet cable or wifi. I chose second. To make pinn connect to wifi I created a `wpa_supplicant.conf` configuration file in the root directory
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
	ssid="ssid"
	psk="pass"
}
```
and added the following flags to `recovery.cmdline`: ```vncshare forcetrigger ssh```. Everything worked fine, the list with all available oses appeared but when i select the ones I need and pressed install, the following error popped up ```"Error writing downloaded file to SD card. SD card or file system may be damaged."``` [(this?)](https://github.com/procount/pinn/blob/master/recovery/mainwindow.cpp#L1705)
I changed several sd cards, used different wifi but nothing seemed to work. Then I restarted the pinn and tried again... and it worked. Everything installed without errors. What I think happens is something does not initialize properly on first boot with preconfigured wpa_supplicant.conf. As a workaround you should reboot you RPi and the second boot should work or you can use ethernet cable. I also think it would be great to have an example of `wpa_supplicant.conf` in the documentation and please add a shutdown button as ssh does not provide such functionality (I tried `halt reboot restart shutdown -h now`).

procount, 2017-10-20:
That's very odd, I've not seen that before and it is part of the original NOOBS behaviour. I shall have to try those exact same steps myself.

Adding an example wpa_supplicant.conf file is a good idea.

The environment does not support the usual shutdown / reboot commands. 
You can, however, use `echo b>/proc/sysrq-trigger` to reboot it.
Other commands probably work too. See for example: https://www.centos.org/docs/5/html/5.1/Deployment_Guide/s3-proc-sys-kernel.html

hurlenko, 2017-10-20:
Thanks, `echo o>/proc/sysrq-trigger` works like a charm. I think it should be mentioned somewhere in the documentation too.

procount, 2017-10-20:
One day I might create some aliases for them so they perform like shutdown etc.

pajo-ojap, 2018-01-27:
I was running into the same issue of not being able to install any OS from pinn_lite.   Again multiple times on a couple different sdcards.  It booted to the pinn screen just fine.  I went back and tried NOOBs_lite and that worked just fine.  I'm glad I ran into this issue here because It made me try  pinn_lite again and after the 1st boot (still failed installs) I rebooted and lo and behold it let me install multiple OS.  I believe the pinn_lite version equates to version 2.45a.  also adding those shutdown/reboot commands to pinn would be a great idea.

procount, 2019-01-15:
reboot is now supported as `rebootp`