procount/pinn #98: Auto boot into os?

Haris1977, 2017-11-19:
Is there any option to auto boot into a specific os (e.g openelec) but only after the 10 second booting time (in order to have the possibility to change os)?

procount, 2017-11-19:
Yes, that's the 'sticky default boot' option. Tick the box next to openelec on the boot selection dialog.

Haris1977, 2017-11-19:
I ve already installed 3 os's now:(

Should i enable "sticky default" by entering the recovery mode? Or do i have to install all os's again?

procount, 2017-11-19:
Just reboot into recovery mode and tick the box next to the OS you want. Thereafter it will boot quicker into that OS unless you press shift to enter recovery mode again.
You only need to reinstall OSes if you want to change them.

Haris1977, 2017-11-19:
yes but..The 10 second booting screen will just "wont appear anymore" with this way..Right?

And i want it to be able to choose an os..

procount, 2017-11-19:
OK, so it's not _exactly_ as you want, but I think there are plenty of options:
1) set an autoboot.txt option  to boot directly into one of the installed OSes. This method hides PINN and any other OSes, so the only way to get back into them is to change or delete autoboot.txt. It is the quickest way to boot, but least flexible.
2) Set an OS to sticky boot. This brings up the recovery splash screen to allow you to press the shift key to select another OS temporarily, otherwise PINN will boot your "sticky" Os after a short time. It is a little slower to boot, but more flexible.
3) The default option is to display the boot selection dialog for 10 seconds (adjustable by menuboottimeout) and then boot the LAST OS. It is the slowest boot method, but quite flexible.
4) Setting forcetrigger forces the PINN recovery menu to display until an option is selected, so it provides an indefinite time before boot, but the most flexibility.
It sounds like you want something like 3) but to boot to the stick/favourite option by default instead of the last OS booted. That's not quite available, but I think there is enough choice without it.

Haris1977, 2017-11-19:
You said in 2 "Set an OS to sticky boot.This brings up the recovery splash screen to allow you to press the shift key to select another OS temporarily, **## otherwise PINN will boot your "sticky" Os after a short time**"

Does this mean that i have 10 second time to press PROGRAMM UP on my tv to change booting ios?

procount, 2017-11-19:
Yes.

Haris1977, 2017-11-19:
Oh nice, thanks, will try it out:)

My thoughts about future updates (in terms of beautification of boot menu)

1) Black background instead of the ugly white one
2) Bigger fonts and bigger square (fonts are very small in big screens) and no cursor in boot menu
3) Give us the possibility to rename (through the shift key) the installed os (eg from openelec_rp1 to OpenELEC Rpi) 

Just my thoughts:)

procount, 2018-01-13:
You can now set the background colour in p2.4.5. Not perfect, but it goes some way. no cursor in boot menu can be specified.
OSes will not be renamed because it is part of how the program works.

Haris1977, 2018-01-14:
Thanks @procount . Any news about bigger fonts/square and programmable cec controls?

MartinX3, 2018-02-08:
You can also mount the SETTINGS partition, as example, if you insert the micro sd card from the raspberry pi into your computer.
There you can remove the sticky option from the config file. :)

closed 2018-05-09 by procount