procount/pinn #99: Error downloading meta file

Xnehaxixh, 2017-11-21:
After setting up PINN for the first time, connected to internet, when I try to network install an os throws me this error,

For LibreElec 
Error downloading meta file
http://mirrors.fe.up.pt/pub/libreelec/noobs/LibreELEC_RPi2/marketing.tar

For Kali Linux 
Error downloading meta file 
http://downloads/sourceforge.net/project/pinn/os/kali2/kali2.png

Any idea why is this happening?

procount, 2017-11-21:
No idea. That's very odd. I managed to download and install all of the OSes at the same time onto an external hard disk with PINN without any issue. Maybe server or internet issues...?
If the problem persists, you could insert a FAT32 formatted USB stick with just an empty /os folder on it. Then `download` all the OSes you want, one by one to the USB stick. Once complete, reboot and install the OSes you want from the USB stick instead of the internet.

Xnehaxixh, 2017-11-22:
Oh, it works now just fine. Maybe it was just server error which was so now fixed.

closed 2017-11-22 by Xnehaxixh