procount/pinn #101: Flavours & customisations on RPI3 - No uncompress?

Jey2k, 2017-11-27:
Hi,
Sorry if my question has already been asked, but I didn't find anything about it.
I just discovered PINN and try to use it to customize a recalbox version.
I downloaded it and put it on USB key in order to install offline.
I made a flavour and gave a ".txt" file..
"Normal" files are well copied
I put some ".tar.xz" files, but these are not uncompressed, and are simply copied to the root of the partition.
If I look to the /tmp/debug file, I have:
`Parsing:  "custojey/themes/albator.tar.xz"
"Copying file"`
[image]
Maybe I made a mistake, but I read the documentation several times and I suppose that's the good way to do it...
Thanks,
Regards

procount, 2017-11-27:
It has been a while since anyone asked about this or I did any work on it, but it's nice to see someone using it.

I suppose you read the documentation on the noobsconfig github? I had to re-read it myself!

It does say that it will decompress tar.xz files listed in the .txt file, but my recollection was that it only decompressed a tar.xz file that was the same name as the fkavour. Files in the .txt file would just be copied. But maybe I changed it.

I will examine the code to see what it is supposed to do....

Jey2k, 2017-11-27:
Yes, I read mainly the noobsconfig for this.... ;)
I suppose this is the code in MultiImageWriteThread::postInstallProcessConfigFile .... And even if it's not one of my languages, I think there is no test based on the extension 
Yes, I had to read several times to understand what was exactly described in noobsconfig documentation.. My understanding is if the tar file (or ".xz") file name is composed of the flavour and partition, it will be automaticaly unzipped....  but because I would like to create several flavours with installation (or not) of some "packages" with a lot of files in it, it would be really great if you could implement back this feature....
Maybe i didn't read your code in the good way, but i think if my line starts with an '@', then it's a kind of include.... If right, this could be another way to split / manage ...
Thank you for your first answer, by the way

procount, 2017-11-27:
Well, it seems I left that feature out of the PINN version compared to noobsconfig, so as I thought, PINN will only copy files listed in the flavour.txt file, not expand them.

Currently, you would have to put all your files in one big flavour.tar.xz file for them to be extracted.

If this is important to you I shall look into putting it in PINN too. Please ket me know

Jey2k, 2017-11-27:
In fact my idea is to propose people to create a custom version of recalbox, choosing in one single file their options (which splash video, which emulated consoles, what kind of games)...and because each of these choices would impact on a lot of files,  yes it would be really useful... So if you have some time to implement it, for sure I would use it..

Thanks

procount, 2017-11-27:
Yes, the '@' is like a nested include.

I'll see what I can do

procount, 2017-11-28:
I have put a copy of pinn-244d.zip on my sourceforge wesbsite in the testing folder.
https://sourceforge.net/projects/pinn/files/testing/
Please see if it now does what you want.

NOTE: If you have {flavour}.tar or {flavour}.tar.xz file in your os folder, it will automatically be _processed_ and expanded to the root folder of the target. Likewise, a {flavour}.txt file will be _processed_ by copying the files in the txt file to the target, rather than copying the {flavour}.txt file itself.
All files listed in the {flavour}.txt file are _copied_ to the target by default.
If you want them to be _processed_ as above instead of _copied_, precede them with an '@' symbol. This will only apply to files ending with a .tar, .tar.xz or .txt file, otherwise it will be ignored.

Jey2k, 2017-11-28:
Great news... I will test this evening... Thanks for your reactivity ... I'll tell you if it works ASAP..

Jey2k, 2017-11-28:
Perfect!! It works great!
I have tested: simple files, tar files et xz files (for non regression) and includes, with simple files, xz files and other includes... All works perfectly... Great job, and a big thanks (you can close Issue)
Several questions/remarks:
1) it seems you don't use the same way to uncompress partition files (by using bsdtar) and these new "@xxx.tar.xz" files (using "xz ... | tar ...") -> normal? no possible code re-use?
2) should I keep this version? will it be automaticaly updated with next release/updates?
3) a simple remark for something that did not seem natural to me: if the include file is in a sub-directory, the list of file may have been found from within this directory.... But in this version, you always have to put the path from the os directory.... It's good for me but I think it could be discussed ;)

Last question (other subject): in NOOBCONFIG documentation, it talks about a "retro.sh" script, for being able to test customisations without having to install whole system again. Is such a feature available with PINN? 

Many thanks

procount, 2017-11-28:
1. It's historical. Noobsconfig was done before PINN. PINN was changed to use bsdtar to install Arch. There's no code resuse. Tar and bsdtar are both external programs. I may change to bsdtar in the released version, but there's not a lot of difference for this use case.
2. Yes. I will update the version for the final release so this version should self-update.
3. Everything is referenced from the same folders regardless of where the include file is stored. Maybe I should write specific documentation for this feature rather than relying on noobsconfig any longer.
4. No, but I thought about it when I was testing... I soon switched my testing to reinstalling lede2 !

closed 2017-11-28 by procount

procount, 2017-11-30:
@Jey2k - I was thinking about your point 3 and relative paths.
In fact I think it would be better if any files referenced in a custom .txt file were relative to the folder in which the custom .txt file was stored in, rather than the original OS folder.
This would mean that you could put a certain set of files in a folder with a custom.txt file and it would be self-contained. It would not matter where it was stored, and indeed it could be referenced from several other include files, maybe even different OSes.

BUT, there may be occasions when you want to use the old way and refer to files in the OS folder. So I added a feature whereby a file beginning with a '/' would refer to the old way of doing it, i.e. relative to the OS's folder.

What do you think? I realise it would mean a change to your custom .txt files, but I think it improves the functionality.
As a first step, you could just prefix all your filenames with '/' and it will work as it does now in 2.4.4d (it would be a good test!). Then in step 2, remove the '/' and folder names for files in the same folder as the custom .txt file (another good test).
If you are interested I will post another beta for you to test.

Jey2k, 2017-11-30:
Hi,
As I told you before, I really think it would be a great way of managing the different flavours .... I love the way you propose things... Yes, starting by "/" , if it represents the OS root is a good idea I think...
Don't worry about my actual work, as I told you I started a few days ago, so not a lot of changes ;)
I could test this new version with pleasure...

Another feature I was thinking about: do you think in these custom files we could make a reference to an http address ? The aim would be to store quite the entire files on a remote server.... and people would DL just a minimal image (=PINN + flavours files) and I'd like to propose them to customize their flavour by commenting on uncommenting some lines, and then when installing it would download only needed files.... One way of doing this could be:
- in custom files, continue working as today for lines NOT beginning by @
- for the others, add the possibility to add a second field which could be an url where to DL the file (because if we already have it in local, no need to DL it)
I suppose I have to think about it a less more, but the concept should be interesting I suppose..

procount, 2017-11-30:
OK I will push another beta up to testing soon.

It will also include a new untested cmdline option `configpath=` which can point to a local folder where the custom files can be stored. This should solve the reverse problem, allowing you to customise an internet hosted OS with a local set of custom files.

I shall have a think about the http option.

procount, 2017-11-30:
I have uploaded pinn-244e.zip to testing.
To use configpath= with a local USB drive, use /tmp/media/sda1 to get to the root of /dev/sda1 etc. USB drive might need a /os folder on it to be recognised.

procount, 2017-11-30:
/tmp/media/sda1 only gets you to the root of your usb stick. But your configpath should be /tmp/media/sda1/os/the_os_i_use.

Not sure if flavours work for internet downloads but you should be able to customise a standard os without a flavour.

Thanks for testing.

Jey2k, 2017-12-01:
No problem for doing some tests, but for now I'm still not comfortable with your new "configpath": in which use case would it be useful? Because in your previous version(=not testing), there already was some kind of "comparison" between internet hosted and USB hosted.... So what kind of help it could be?

I can give you the ways I'd like to finally use your software, and then you could see what I would need and what could be actually missing:
- Actually, i downloaded OS on my USB card, and added all my flavours and customizations -> good for offline use, but 1) Need to manually update OS if the supplier release an update 2) Need for "customers" to update their USB key if I make some new features/flavours 3) People need to download the whole configuration with all customized files, even if they will use a "light flavour"

- For me, next step would be to use a internet OS, but a local flavour -> then, if my internet supplier (Recalbox, for me) release an OS update, I don't need to change my USB key... So my "customers" won't need an update/download again the content of their USB key.

- Ultimate step would be, in my opinion,  a full online (and only if needed) installation AND customization, with flavours and needed files stored on a webserver. The behaviour would be: 'ok, for this flavour, I need the "xxx_root.tar", it's not present on the USB key, so I will DL it', but next time, I could DL a offline version of a flavour -> In this case, the minimal "customer download" would only be a PINN-lite installation + a config file .... and the whole files would be hosted on my personal webserver.... like the supplier for their OS in fact..... so as soon as I publish an update, all my "customers" using online installation would have the last version....

Not sure it's clear.... my English is not really the best :/

procount, 2017-12-01:
Configpath allows for internet os but local flavour, which is not currently supported.

You're English is fine. I am working towards a full solution step by step.... Next version should be more improved.

Leobaillard, 2018-06-18:
Hi! I'm hopping on the train a bit late, but I'm very much interested by that feature.

I'm trying to create a silent network installation of Recalbox which also includes a customization to enable by default the GPIO controller by modifying `recalbox.conf`.

From what I got from this thread, it looks like this might be possible? I'd like things to happen like this:
- Pi boots in PINN
- PINN checks for Internet connection (most likely, Ethernet)
- If connection is detected, proceed to the silent install using the network and `os_list_v3.json` populated only with Recalbox
- Recalbox installs from Internet
- After install is completed, `recalbox.conf` is copied from the customization folder to the OS partition, rewriting the file (ideally, it would be better to do a search/replace on the specific configuration parameters, but hey, I'll take what I get)
- Pi reboots into Recalbox fully configured and with the GPIO controller enabled

I already have modified a local version of NOOBS to enable the silent network install and I'm trying to do the same using PINN and the customizations.

procount, 2018-06-18:
The customizations are an update in progress at the moment (in order to fully support customisations during the network install) and are therefore subject to change. I hope to get back to them soon; I think the release of the 3B+ distracted me...

Leobaillard, 2018-06-18:
Ok. Might I be of assistance in completing this functionnality? What's missing for it to work on network installs?

If I can't help, do you have an estimate as to when it'll be available? Thanks!

procount, 2018-06-18:
Well, to be honest, I can't quite remember where I got to! 
I think it was mostly there, but it needs thorough testing. So if you are in a position to create your own repository, any testing of the various combinations of configurations would be great. i.e. 
1) OS on local SD card/USB, customisations on local SD card/USB
2) OS on remote repository , customisations on local SD card/USB
3) OS on local SD card/USB,  customisations on remote repository
4) OS on remote repository,  customisations on remote repository
And all the above combinations with the customisation files to be installed at local file addresses or remote http: urls.
I've still got some things in hand at the moment, but I expect I will come back to customisations aftert hose are complete.
(If you are willing to post your local NOOBS network changes, I can see if it is worthwhile to put them in PINN too)

Leobaillard, 2018-06-18:
Nice!

I can try some of these combinations and report back. My goal is to have the customisations on the SD card at the moment, but I could also try USB.
I will try with the `configpath` cmdline option, but I don't know what the path for the SD root is.

Regarding the modifications I made to NOOBS, I don't know if they are worthy, but you can find them here: https://github.com/TVL-Communication/noobs/commit/10b5e42712cb32118a322c9348b460772fda27c3
It basically adds a cmdline argument `silentnetworkinstall` that triggers an install of a single OS config when a network connection is established. Never mind the password field bit, this was done to accomodate a client's request.

procount, 2018-06-18:
> I will try with the configpath cmdline option, but I don't know what the path for the SD root is

I think it would be just /mnt (followed by /os/your-os)

procount, 2018-06-19:
See https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&p=1258369#p1258369 - It seems I haven't quite finished the implementation of flavours.

Leobaillard, 2018-06-19:
I haven't tested flavours but replacing configuration files from files stored in a directory at the root of the SD card works flawlessly! I was able to silently install recalbox from the Internet and pre-configure it with language and gamepad configuration.