procount/pinn #105: Arch download still references remote files

procount, 2017-12-07:
From symbios24 on sourceforge review:

About the ArchLinux both versions Arch1 and Arch2 i have downloaded all the files required for an offline installation from the sd card and it ask for a file/s between the installation from the internet, as i did my research i found out that the partitions.json file includes in the tarball section this two entry`s ---githubusercontent.com/procount/pinn-os/master/os/Arch2/boot.tar.xz--- ---archlinuxarm.org/os/ArchLinuxARM-rpi-2-latest.tar.gz--- So it requires an internet connection even you you have downloaded all the necessary files for offline installation can you fix that please? as i see now the archlinux os from the OS folder is not there anymore,i will appreciate if you can make both archlinux versions totally offline like the other os Thank you

jkw, 2017-12-10:
I did not even get Arch 1 nor 2 in the list even WITH a connection. :|

procount, 2017-12-10:
Did you look on the "minimal" tab group?

jkw, 2017-12-12:
I managed to get it work in latest update. 👍

symbios24, 2017-12-30:
Has this issue fixed yet?

symbios24, 2017-12-30:
Hello, has the issue of downloading the arch files from internet fixed?

procount, 2017-12-30:
Not yet. Arch was done differently because it is a "rolling release" and continually updated. If I cache a copy it will soon get out of date. I will look into it in the New Year

symbios24, 2017-12-30:
Thanks Merry Christmas

procount, 2017-12-30:
I must have forgotten I'd fixed this because I just tried it on Arch2 and it worked fine (I've not tested Arch1)
Please re-download Arch2 with the latest PINN version and then try to install it. You should disconnect the PI from the internet when installing Arch to ensure it is getting it form your local source.

procount, 2018-01-08:
Assuming this is now fixed

closed 2018-01-08 by procount