procount/pinn #106: Manage pull request

bovirus, 2017-12-07:
@procount 

There are 4 pull requests about language updates (2 quite old - 2 quite new).

Please process as soon as possible the pullup request, expeccially about language updates, because otherwise the translator have to start again to revise the translation.

Thanks.

procount, 2017-12-07:
I'll try my best to do them soon.  very busy at the moment, but I appreciate all the hard work of the translators.

bovirus, 2017-12-07:
@procount 

Please use highest priority for language updates.
Probably the 2 old requests are too old because there were many changes to the languages.
But the recent two requests (it/fr) should be applied as soon as.

Thanks.

procount, 2017-12-08:
merged

closed 2017-12-08 by procount