procount/pinn #107: Headless raspbian lite install with ssh

j0hnsmith, 2017-12-09:
Is there a way to install raspbian lite with ssh support from first boot with pinn lite? The [enable ssh in raspbian](https://github.com/procount/pinn/blob/master/README_PINN.md#enabling-ssh-for-raspbian) from the readme doesn't work with an OS installed via the network.

procount, 2017-12-09:
I think you might have done something wrong, because I just tested it and ssh worked fine.
I installed PINN to the SD card and created an empty ssh file in PINN's recovery partition (/dev/mmcblk0p1). In PINN I enabled wifi support and installed Raspbian-Lite from the internet. When I booted Raspbian-Lite, it connected to the network over wifi and I was able to ssh straight into it.

Note that this is a feature of Raspbian's and Raspbian Lite's partition_setup.sh and won't necessarily work with all OSes.

I suggest you try again and if you're still having problems, see the troubleshooting section of the wiki and post a copy of /tmp/debug and the detailed steps you took.

j0hnsmith, 2017-12-09:
I'm in PINN via vnc but I can't get to `/dev/mmcblk0p1`. I'm on a mac, tried sending all the control/option/cmd/shift + F2 combinations in 2 different vnc clients. Is there another way to access the recovery partition?

procount, 2017-12-09:
No, you can't do that via VNC.
The easiest way to create the ssh file is when you originally put the SD card in your mac and copied the pinn-lite files to it. Just add an empty ssh file at that time. You can still do that now, by just removing the SD card and putting it in your mac.
But if you don't want to remove your SD card from the RPi again, you can create it using SSH.
You must have first enabled ssh in PINN by adding `ssh` to recovery.cmdline. Again, this can be done by putting the sd card in your mac and editing it (but if you're going to do that you may as well add the ssh file at the same time). 
Alternatively, if you have the latest PINN version, you can go the maintenance menu, highlight PINN and select the Edit Config file option. From there you can add `ssh` to recovery.cmdline and then reboot PINN. ssh should now be enabled.

To log in via ssh use the username of `root` and password of `raspberry`.
The /dev/mmcblk0p1 partition is mounted on /mnt, but it is normally read-only. To write the ssh file to it you have to:

```
mount -o remount,rw /mnt
cd /mnt
touch ssh
sync
mount -o remount,ro /mnt
```
Now you can use the new re-install option to reinstall Raspbian-Lite and ssh should now be enabled.

j0hnsmith, 2017-12-11:
Still not working for me, here's the steps I'm doing

* download pinn lite from sourceforge
* add empty `pinn-lite/ssh` file
* copy `wpa_supplicant.conf` into `pinn-lite` dir
* add `vncinstall ssh` to `recovery.cmdline`
* format sd card and copy all files under `pinn-lite` dir to it
* put sd card in Raspberry Pi and connect via vnc
* install Raspbian Lite
* press ok to restart the pi

Still didn't work so I enabled ssh for pinn, here's what I see

```sh
# mount
/dev/ram0 on / type squashfs (ro,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=98936k,nr_inodes=24734,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
/dev/mmcblk0p5 on /settings type ext4 (ro,relatime,data=ordered)
/dev/mmcblk0p1 on /mnt type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
# cd /mnt
# find . -name ssh
./ssh
```

So the empty ssh file is there. I can see the raspberry pi on the network, just can't ssh into it.  One error I do get the first time I try to install the new os is this

[image]

however when I restart the pi and install again it goes away and the install proceeds as expected (as far as I can tell).

I feel like I'm making a really simple mistake. I've tried both Raspbian and Raspbian Lite.

When I've figured this out I'll add detailed 'headless' steps to the readme.

j0hnsmith, 2017-12-11:
So I dug out a mouse and keyboard, turns out my `wpa_supplicant.conf` was incorrect, strangely my router was showing me an IP for the pi, but `ifconfig` wasn't. `/etc/wpa_supplicant/wpa_supplicant.conf` was empty!

Invalid file, not sure where I got it from

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=wheel
network={
    ...
}
network={
    ...
} 
```

Valid file

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ...
}
network={
    ...
}

```

I'll write up some detailed steps and make a PR. Thanks for your help, sorry for wasting your time.

closed 2017-12-11 by j0hnsmith

procount, 2017-12-11:
No problem. 
Glad you identified your issue.