procount/pinn #108: Migrate from NOOBS preserve installed operating systems

Jip-Hop, 2017-12-27:
Is there a way to install PINN without losing the operating systems installed with NOOBS? I was hoping I'd be able to overwrite NOOBS and continue to use PINN because it has HDMI-CEC support when choosing the OS to boot. But I'm not looking forward to reinstalling...

procount, 2017-12-28:
Yes. But first make a backup of your SD card (just in case)!

Jip-Hop, 2017-12-28:
That didn't go so well. I found this line `runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json` in `recovery.cmdline`. I removed `runinstaller` and left the rest. Now it has the same content as the `recovery.cmdline.new` file.

Then I copied all files from pinn-lite.zip over the NOOBS files in the following partitions: root partition (has no name), boot, boot-rbp2.

When I boot my Raspberry Pi 3 I'm presented with PINN OS selection screen. But it's a bootloop. When the time to select an OS runs out, it's supposed to boot into the previously selected OS. Instead it boots back into the PINN OS selection interface. When I manually select an OS to boot, I get a bootloop too but I no longer get the PINN OS selection interface.

Also I noticed HDMI-CEC wasn't working in the OS selection screen (Philips TV). This was actually the reason to go with PINN. HDMI-CEC works when booting in Kodi so I'm sure my TV is capable of it.

I did make a backup so I can restore.

procount, 2017-12-29:
You only need to copy the pinn-lite files onto the first NOOBS recovery partition (/dev/mmcblk0p1). The others should stay as they were. Otherwise, I expect when you boot into your selected OS, you will boot into another copy of PINN again.

I suggest you restore your backup and try again, just with the first partition (as if you were installing PINN for the first time). don't forget to check `runinstaller` is removed before booting again.

procount, 2018-05-09:
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=212452

closed 2018-05-09 by procount