procount/pinn #109: More OS

symbios24, 2017-12-30:
Hello there, Can you Add/update this Operating Systems?

RetroPie to v4.3
DietPi
KanoOS (http://dev.kano.me/public/Kanux-Beta-latest-release_noobs.zip)
RTAndroid (last version 20170404)
Openplotter (https://mega.nz/#!aMgTDZIR!Reotv8jDECzVKjBUDOXncW63hAzWiEADefkZBH0EJys)
AmiBerry
MotioEyeOS
ChromiumOS https://sourceforge.net/projects/rpi-chromium-boot/files/

Thanks

procount, 2017-12-30:
I will look into these in the New Year.

solarsootysmudge, 2018-01-05:
Needs download links for Ubuntu MATE added

procount, 2018-01-05:
Some OSes are not compatible with PINN/NOOBS usb boot so are hidden :(

procount, 2018-01-05:
Pinging @matthuisman for KanoOS and Chromium, but I think Flint_Os is a suitable alternative to Chromium isn't it?

matthuisman, 2018-01-06:
@procount  

Happy New Year :)

Have just updated KanoOS to lastest Beta 3.14.0.
Way easier now as they have a NOOBS package.
I suspect they will try to get into official list eventually.

Not sure about that Chromium.
I'd rather stick with Flint OS as it seems pretty stable.

procount, 2018-01-13:
I've updated the OP with the current status of the OSes.

RTAndroid now requires an email registration to download, It also seems the last one as they have changed the name to something else. I have added Lineage OS instead.

Flint OS is a suitable alternative to Chromium so that will not be added.

symbios24, 2018-01-14:
Hello thanks for the updates,is it possible to add each os in a zip archive? since downloading one by one the files and organizing them after can take a while.i think is easier for you also, you upload one file only for each os and not 20 or 25 separate.

The links i had give you for openplotter and kano os are already converted noobs archives, if you can add them anytime later will be great.

Thanks

________________________________

procount, 2018-01-14:
Kano has already been updated by Matt

fscii, 2018-05-13:
Is there any way for the end user to simply add an OS to the menI?  Since we are to download the OS manually, why can’t I just edit something to say hey add this OS I just compiled myself last night etc?  Or is that there and I couldn’t see it?
That’s how I found this hoping noobs could do that, then seeing that beery boot is like 1743 steps to do it.

I’d its already here then please point the noob in the right direction but if not please add this feature

procount, 2018-05-13:
Yes of course there is.
The main source of documentation can be found at https://github.com/procount/pinn/blob/master/README_PINN.md

In particular you may like to look at https://github.com/procount/pinn/blob/master/README_PINN.md#installing-oses which explains the many ways to install the OSes. 

But you have to format your OS in a particular way in order for it to be installed by PINN, which is exactly the same way as for NOOBS. You can find instructions here -> https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version

If you already have an SD card with your perfect OS on it, then the Wiki has an article on how to turn it into an installable OS here ->https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN

fscii, 2018-05-14:
Thank you very much !