procount/pinn #112: cant install drivers

678GUY, 2018-01-02:
so i have an LCD screen for my rpi3 but i cant install them because the recovery shell dosent work with git or with my usb drive and if i install the drive inside an os it reboots and PINN wont boot after that so there is a problom
driver: https://github.com/goodtft/LCD-show
and i install the file called LCD5

procount, 2018-01-03:
Most USB drives should work in PINN's recovery shell, but you must mount them manually.
e.g. 
```
mkdir /tmp/usb
mount /dev/sda1 /tmp/usb
```
Do you have a link to the LCD screen product?
Looking at the LCD5 script, it looks like it has a HDMI interface, so creating a suitable config.txt file should make the display usable.  The touchscreen interface is another matter, though....

678GUY, 2018-01-03:
the amazon link
https://www.amazon.com/gp/product/B01LZN54R0/ref=od_aui_detailpages00?ie=UTF8&psc=1
ok so i mounted it but everything i do dosent seem to run it its or cant open or premission denied or couldent find the file or somthing like that also i dont care about the touch screen but the driver makes the screen the size it needs to be there is a black chunk at the side of the screen

procount, 2018-01-04:
Please try copying `config-5.txt` from the LCD_Show github /boot folder to the 1st partition of your SD card where PINN is stored (/dev/mmcblk0p1) and rename it to `config.txt`. You can do this on a Windows PC if you are struggling with PINN's recovery console. That should be sufficient to get the display the right size in PINN.

The touchscreen would need additional drivers and support adding to PINN. I may add support for it in the future if it is useful when I have time, if someone is willing to test it for me, but it would be low priority.

678GUY, 2018-01-04:
thank you very much it worked and also i would be happy to test it for you if you need it

WHITESKULL1, 2018-01-20:
hi just came across your issue, Im trying to install some kind of software that will help get me started I'm a newb , any freeware you recommend i have a new mac air I don't know alot to get the software i downloaded setup. thanks

WHITESKULL1, 2018-02-16:
> what software do you need?

Hi sorry for the late response kali Linux. Thanks for your help

678GUY, 2018-02-16:
so you want tools to install kali linux on a raspberry pi?